dcnieho/Titta #4: Error with getSDKVersion

iandol, 2019-08-13:
Hi, after pulling the latest changes I get this:

```
>> readme
PTB-INFO: Multi-display setup in explicit multi-display mode detected. Using the following mapping:
PTB-INFO: Screen 0 corresponds to the full Windows desktop area. Useful for stereo presentations in stereomode=4 ...
PTB-INFO: Screen 1 corresponds to the display area of the monitor with the Windows-internal name \\.\DISPLAY1 ...
PTB-INFO: Screen 2 corresponds to the display area of the monitor with the Windows-internal name \\.\DISPLAY2 ...

PTB-INFO: Your version of Matlab 64-Bit is global system DPI aware. On Windows-8 or later, fullscreen onscreen windows will only work 
PTB-INFO: properly timing-wise when displayed on displays with the same pixel density as your systems primary display monitor.
PTB-INFO: For your multi-display setup the stimulus display monitor must have a DPI of (96, 96), matching that of
PTB-INFO: your primary display monitor. Ideally you will only display on the primary display in the first place.
PTB-INFO: Displaying on anything with a different DPI will cause mysterious visual timing problems, sync failures etc.
PTB-INFO: Read 'help RetinaDisplay' for more info on this topic.
Error using TobiiBuffer_matlab
Unrecognized action (not in actionTypeMap): getSDKVersion

Error in TobiiBuffer/cppmethod (line 42)
            [varargout{1:nargout}] = this.mexClassWrapperFnc(methodName, this.instanceHandle, varargin{:});

Error in TobiiBuffer/getSDKVersion (line 76)
            SDKVersion = this.cppmethod('getSDKVersion');

Error in Titta/init (line 341)
            out.systemInfo.SDKversion.tobiiBuffer_C = obj.buffer.getSDKVersion();

Error in readme (line 62)
    EThndl.init();
 ```

dcnieho, 2019-08-13:
Thanks for the reports! all are very welcome. I wrote some new code at home
yesterday, didn't test or recompile the mex file yet, hence the error
(misleading commit message i used i see). Use one revision earlier.

I plan to completely get rid of Tobii's matlab SDK. Its a lot of extra
stuff to bring in, especially if i have my own mex already that could
simply wrap some more of the C SDK. Some more breakage may ensue over the
next few days. (Perhaps i should move to using feature branches!)

iandol, 2019-08-13:
OK no problems, I'll just checkout a working commit and wait...

dcnieho, 2019-08-13:
Hi Ian,

Master should work again. There is also a branch now where i have
completely replaced Tobii's matlab SDK, in case you want to try already.
Its working for me :)

Cheers,
Dee

iandol, 2019-08-14:
Yes, I can confirm it's working ;-)

closed 2019-08-14 by iandol

iandol, 2019-08-14:
Hi @dcnieho ,

Out of curiosity, are there any advantages to removing the Tobii SDK dependency, or is is just to "simplify" the interface?

dcnieho, 2019-08-14:
Glads its working, thanks for trying! I wrote a mail yesterday to your
gmail, but realize you may not be able to access that. here it is, edited
further

--------

Hey Ian!

Thanks a lot for using Titta and noting the issues you have found.

I'm glad you are pushing me for Linux support. Will figure that out.
Let me first fully finish the Windows version, which i should manage
this week still. The removeTobiiMATLABSDK branch already has a version
of the code that fully replaces Tobii's MATLAB SDK. As you also noted,
i'm not a fan of some of their decisions, nor have i ever gotten it to work
with Octave at
all (which i also hope to support).
May i ask you later to try out that branch (just a quick test) before
i merge it to master? And same for a Linux version?

The reason to completely replace the SDK is indeed that its not much work
to replace the last bits from what i already have, reducing dependencies
and simplfying. And this way, i can use an interface i like (e.g. something
that returns doubles instead of singles), no MATLAB classes for what is
just POD, few things like that. The big benefits of my mex file, for me,
are of course the peek/consume interface, and the asynchronous interface to
the calibration functions.

Cheers and thanks,
Dee

iandol, 2019-08-14:
Thanks Dee! Yes, I'm happy to test whenever...

Best, Ian