dcnieho/Titta #5: Floating Point Colour as a default?

iandol, 2019-08-14:
Hi Dee,

I always use floating point colour in PTB, and I notice all the colours in Titta are 8bit. I can manually edit all the options returned by  titta.getDefaults, but this is a lot of boilerplate code. I wondered whether adding  an argument to the static method to return values as floating point/8bit would be worth adding? 

The simplest thing is just a divisor, e.g. [255 127 0] / 255 - there is a rounding precision error but for the UI this is not an issue...

iandol, 2019-08-14:
Ah, I just saw `qFloatColorRange`, it seems you test the screen on init and if it is floating point then divide by 255 every time you use a colour. Slightly less efficient (you have to call `getColorForWindow` every time a colour is used), but doubt it would lose more than a few microseconds here and there...

dcnieho, 2019-08-14:
Hi Ian,

The code should automatically detect when the framebuffer is using floating
point color and convert the 8bit colors to 0--1 upon drawing (so not
reflected in the settings). Is that not working, or do you mean that you
would like to provide floating point colors as input?

Cheers,
Dee

iandol, 2019-08-14:
I hadn't got as far as testing with a screen yet, I was reading through code first... ;-)

Yes, on testing the auto-conversion is working fine. As I mentioned above, it would be ever so slightly more efficient to do the conversion at the getDefaults stage (only happens once), rather than continually during drawing via `getColorForWindow`. But this is super minor...

closed 2019-08-14 by iandol

dcnieho, 2019-08-14:
Thanks for noticing that. Don't read all the code (unless you like), it
gets a bit hairy and some code isn't nice (e.g. that button class is shit
but works).

I wouldn't want to do that in getdefaults yet, that would require a window
to already be open at that time. Performance impact should be non-existent
as its a simple division. Perhaps i could cache the results upon function
entry, would make the code cleaner too. Yeah, I'll do that. Thanks!
As said, keep em coming, your comments :)