dcnieho/Titta #9: Drift Correction?

iandol, 2019-08-18:
Eyelink offers a manual drift correction, basically a very simple offset applied from a single point. I wonder whether you think it would be a method one could consider for Titta's interface? Perhaps there's a reason Tobii doesn't offer drift correction, but it is useful as slow drift seems to be a problem so far with the spectrum pro. It could be an offset that was only applied to peek()ed data, keeping the stored data untouched. Stored messages detailing drift correction time/value could be used for later manipulation if so desired.

dcnieho, 2019-08-25:
Hi Ian,

Thanks for the suggestion. This is also a feature that i think should be
implemented by the user on top of functionality that Titta provides. I want
to keep Titta lean.

Cheers,
Dee

closed 2019-08-25 by dcnieho

iandol, 2019-08-25:
Hi Dee, OK I'll come up with something in my own interface.  Cheers, Ian