dcnieho/Titta #11: Cannot set frequency

iandol, 2019-12-16:
Trying to run smoothPursuitDemo I see this error (which I also see running my own code):

```
>> smoothPursuitDemo
Error using Titta/init (line 415)
Titta: Error setting tracker sampling frequency to 600. Possible tracking frequencies for this Tobii Pro Spectrum are [1200, 600,
300, 150, 120, 60].
Raw error info:
Error using TobiiMex_matlab
Unrecognized action (not in actionTypeMap): setFrequency

Error in TobiiMex/cppmethod (line 74)
            [varargout{1:nargout}] = this.mexClassWrapperFnc(methodName, this.instanceHandle, varargin{:});

Error in TobiiMex/set.frequency (line 245)
                this.cppmethod('setFrequency',double(frequency));

Error in Titta/init (line 411)
                obj.buffer.frequency = obj.settings.freq;

Error in smoothPursuitDemo (line 71)
    EThndl.init();

Error in smoothPursuitDemo (line 71)
    EThndl.init(); 
```

dcnieho, 2019-12-16:
Hey! Sorry, didn't get to recompile the Linux mex file yet. This reminds me
i should be doing this in a separate branch so as to never break master.
Will be fixed in a few hours!

iandol, 2019-12-16:
No worries, I can roll back in git until it is fixed...

dcnieho, 2019-12-16:
Sorry again, rebuilt now.
By the way, do you use gaze contingent stimuli (or online gaze data for something)?

closed 2019-12-16 by dcnieho

iandol, 2019-12-16:
All of my use is online, as we almost always need to ensure maintained fixation and use eye saccades to target to denote decisions within the trial. We also have some saccade-contigent stimulus change paradigms. That is why your buffer.peekN is so valuable, as Tobii's SDK completely fails for these...

dcnieho, 2019-12-16:
Cool to hear! I am afraid i also have a note of caution then. I have last week been testing temporal performance of Titta. It turns out that there appears to be a bug somewhere in the Tobii SDK that causes no samples to arrive every now and then for intervals of 100 to 160 ms. Its not in Titta as i get the same with completely minimal C++ code using directly their SDK.
Here's a plot of what i mean:
[image]
This is a plot of timestamp data generated with this code:
```c++
#include <tobii_research.h>
#include <tobii_research_eyetracker.h>
#include <tobii_research_streams.h>
#include <vector>
#include <iostream>

#define WIN32_LEAN_AND_MEAN
#define NOMINMAX
#include <windows.h>

std::vector<double> _gazeReceiveTs;
std::vector<TobiiResearchGazeData> _gaze;

double PCFreq = 0.0;
__int64 CounterStart = 0;

void TobiiGazeCallback(TobiiResearchGazeData* gaze_data_, void* user_data)
{
    LARGE_INTEGER li;
    QueryPerformanceCounter(&li);
    _gazeReceiveTs.push_back(double(li.QuadPart - CounterStart) / PCFreq);

    _gaze.push_back(*gaze_data_);

    //QueryPerformanceCounter(&li);
    //_gazeReceiveTs.push_back(double(li.QuadPart - CounterStart) / PCFreq);
}

int main(int argc, char* argv[])
{
    TobiiResearchEyeTracker* et;
    TobiiResearchStatus status = tobii_research_get_eyetracker("tet-tcp://169.254.9.134", &et);

    _gazeReceiveTs.clear();
    _gazeReceiveTs.reserve(100000);
    LARGE_INTEGER li;
    QueryPerformanceFrequency(&li);
    PCFreq = double(li.QuadPart) / 1000.0;

    QueryPerformanceCounter(&li);
    CounterStart = li.QuadPart;

    _gaze.reserve(100000);
    tobii_research_subscribe_to_gaze_data(et, TobiiGazeCallback, nullptr);

    Sleep(6000);

    tobii_research_unsubscribe_from_gaze_data(et, TobiiGazeCallback);
    Sleep(100);

    for (auto& i : _gazeReceiveTs)
        std::cout << i << std::endl;

    return 0;
}
```

What the code basically does is register a callback (thats the way to get gaze data online), and then take a timestamp every time data arrives. I then plot the diff of these timestamps, showing the interval between arrival of consecutive samples. As you see, there are big peaks there. And a steady trickle at about 1.67 ms between samples (I was running at 600 Hz) is rare, thats those plateaus just above 0 in some parts of the plots.

So no samples are lost this way, they eventually arrive, but for online data this can be a real issue.
I have reported it of course, lets see what they say. Will keep you updated

iandol, 2019-12-18:
@dcnieho -- yikes, thanks for reporting this to Tobii; I currently take the last peeked value and don't check it is new, so probably I wont see this issue in Titta? if I add a new sample check then I should see it, and I can post a case on this as well as it is a serious issue...

dcnieho, 2019-12-19:
yes, as long as there is something in the buffer, you'll be able to peek.
Latest can be quite old, so you'd have to check for it being stale. I would
be curious to hear if you see this on Linux or its only a Windows issue. It
would indeed be good if you report it too, i guess the more people having
problems with it, the more chance it gets fixed soon. I hope.

iandol, 2019-12-19:
As I'm not entirely sure what I'd need to do to recompile the C++ code for Linux, I will check from MATLAB....

iandol, 2019-12-19:
OK, this is easily confirmed from within MATLAB:

```matlab
for i = 1 : nSamples
	td = me.tobii.buffer.peekN('gaze',1);
	if ~isempty(td)
		ti(i) = double(td.systemTimeStamp); 
		tx(i) = td.left.gazePoint.onDisplayArea(1);
	end
	tj(i) = WaitSecs(sampleInterval);
end
```

I'm using a sample interval run via `WaitSecs` and using its return times as well as the Tobii system timestamp:

[image]

10ms sample interval 150Hz (should be ~6ms resolution) tobii frequency. You can't even see the PTB timestamps, this confirms it it not a system problem, note the Std. Dev. of the PTB timestamps is 238X smaller than the Tobii SD! 

[image]

600Hz and still dropouts. One interesting feature in both plots is the inter-spike intervals are very regular, basically around 200 samples (10ms per sample so around every 2 seconds)... It looks like they are doing some other function every ~2 secs in their code causing these online hangs.

iandol, 2019-12-19:
Submitted to Tobii, case 00477220.

dcnieho, 2019-12-19:
Thanks for confirming and submitting! The problem looks very similar as on
the Windows side, just here it is less regularly spaced in time, and the
spikes range from 100 to 160 ms instead of just plain roughly 100 ms. I'll
forward your post to the SDK guys as well, hope that helps a bit.

What is your X-axis by the way, sample number? Ah, i see your loop is
running at 10 ms, while the eye tracker underlying is at either 150 or 600
hz. Interesting to see its so regular in time indeed.

Note also that, perhaps useful info: for 150, 300 and 600 Hz mode, the
Spectrum runs at 600 Hz internally. I think it then averages 4 samples to
output one 150Hz sample, or 2 samples to output a 300 Hz sample. I say i
think as i am not 100% sure it doesn't just drop three and output every
fourth.

iandol, 2019-12-20:
If I use 1200Hz the SD does go down a little bit, but the interval still looks similar:
[image]

There are a couple of PTB noise spikes from a background task in that example, but they are not correlated to the Tobii spikes.

dcnieho, 2019-12-20:
Yes, that confirms my experience. Dropping all the way down to 120 also
doesn't solve the issue

iandol, 2020-01-08:
@dcnieho -- I was told today that this has been fixed in a newer SDK that you need to update to. But I didn't see that on the SDK download page?

dcnieho, 2020-01-08:
I'm afraid that info must be incorrect. I am using the latest SDK, and am
in contact with devs at Tobii--its an open case as far as i know.

iandol, 2020-01-09:
Yes, it was a miscommunication and they told me their R&D found a solution that will be in the next SDK update...

commit 21278e58 (2019-12-16): rebuilt linux mex file. Closes #11