dcnieho/Titta #13: Minor: fixation points misplaced when two screens not the same dimensions

iandol, 2020-01-17:
Hi Dee,
On this line you use `w`:

https://github.com/dcnieho/Titta/blob/master/Titta.m#L1259

which is set in a previous loop (line #1173) to be the second screen, when you are drawing these points to the first screen. Easiest way to test is set second screen to be windowed (i.e. `[0 0 1000 1000]`), and you'll see the drawing on the first screen constrained to [1000 1000]...

dcnieho, 2020-01-20:
Thanks for reporting!

iandol, 2020-01-21:
Thanks for fixing! 👍

iandol, 2020-03-02:
@dcnieho -- I notice there are still some problems when you draw the calibration results on the second display if it isn't the same size:

[image]

dcnieho, 2020-03-02:
Hmm, I'm pretty sure that this is "by design" (the logic of that screen is
to have everything 1:1, so the visualized validation errors are directly
interpretable), but now think that that makes little sense when working
with a second monitor, very little. Will fix, probably next week.

dcnieho, 2020-03-11:
Pff, this was thorny. But now hopefully fixed properly. Let me know @iandol if you have a chance to test!

commit 0b2ad71f (2020-01-20): setup display: in several places, the wrong screen information was used for drawing:
- wrong fixation point locations on main screen
- wrong head and ref circle position on operator screen
- not completely ok logic for positioning setup message on both screens
Fixed.
Closes #13.
commit 8957362d (2020-03-11): - second operator screen/window of different resolution than participant screen now fully supported. A scaled down or scaled up version of the participant screen will be shown on it.
- all mouse coordinates used in Titta code are now relative to the screen the buttons are on.
- Replaced used of RemapMouse, as it had undesirable side effects with multiple screens.
Closes #13.
closed 2020-03-11 by dcnieho