dcnieho/Titta #14: Error message while running the "readmeTwoScreens" demo

GJeanChar, 2020-02-10:
Hi, 
I'm new to tobii and I was thrilled when I discovered your toolbox, I will working with infant and I need to implement a calibration script, that's why your "readmeTwoScreens" demo is perfect but when I run it, I can perform the calibration but afterwards I have this error message :

Error using dot
Too many input arguments.

Error in Titta>AngleBetweenVectors (line 2835)
angle = atan2(sqrt(sum(cross(a,b,1).^2,1)),dot(a,b,1))*180/pi;

Error in Titta/getDataQuality (line 2154)
            angs2D      = AngleBetweenVectors(vecToPoint,gazeVec);

Error in Titta/ProcessValData (line 2117)
                    val.quality(p).left  = obj.getDataQuality(val.gazeData(p).left ,val.pointPos(p,2:3));

Error in Titta/DoCalAndVal (line 1659)
                out.val{iVal} = obj.ProcessValData(out.val{iVal});

Error in Titta/calibrate (line 581)
                    out.attempt{kCal} = obj.DoCalAndVal(wpnt,kCal,out.attempt{kCal});

Error in readmeTwoScreens (line 126)
        tobii.calVal{1}         = EThndl.calibrate([wpntP wpntO]);

I really don't get what went wrong here. Could you enlighten me ?
Fiy, I'm using the Tobii Pro X3-120 EPU.
Best regards
Geraldine

dcnieho, 2020-02-10:
Hi Geraldine,

To help i need some more info first:
What version of matlab are you using? Do you get the same error with the
plain readme.m?

Cheers,
Dee

GJeanChar, 2020-02-10:
Hi Dee,
Thanks for the quick reply!
I'm using 2018b and yes I got the same error with "readme"
Best regards
Geraldine

dcnieho, 2020-02-10:
Hi Geraldine,

This is a really puzzling error. Can you run the following lines on the
command prompt and provide me the output:
which atan2
which sqrt
which sum
which cross
which dot

Thanks,
Dee

GJeanChar, 2020-02-10:
Hi, 
well, thanks to your directions I found the reason behind the error : someone add a toolbox and the dot.m used wasn't the one from Matlab, I've changed the path and everything is fine now! 
Thanks a lot.
I do have another question though, I've created my own stimuli for the calibration (a growing red circle with a sound whenever the position change) and I would like to know how to use these stimuli with your way of calibrating, I've look inside the "AnimatedCalibrationDisplay" but I'm kind of lost with the nested functions there. 
Cheers
G.

dcnieho, 2020-02-10:
Hi G,

Ah, great you could resolve it!

That file doesn't have nested functions, but is a classdef (so you know
what to google if you want to know more).
But you don't have to make something like that. In the end, a normal
function with the same signature as doDraw in AnimatedCalibrationDisplay
will do the trick. You see in the demos how to tell Titta what function to
call. make a function_handle to your function, and you should be good to
go. For the drawCmds, you can simply ignore the ones you don't care about.
You probably only care about 'draw' and 'new'. when new is received, the
position has changed, so you can play your sound. When 'draw' is received,
it is a regular draw, so you can draw it slightly bigger than last time
(that does require you to have a way to know what size you drew last).

Furthermore key for the calibration function is that it returns whether the
display is in a state where the user can press the accept button (or the
system can auto accept). Thats there since you don't want calibration data
to be collected if the displayed point is not yet at the requested location
(c.f. the animations in my demo).

There may be another way: you can use the AnimatedCalibrationDisplay
itself, configure it (see demo for color changes and such, and as you see
in the list of properties, its also possible to switch off the various
animations), and use it. You can then register another function
for settings.cal.pointNotifyFunction and settings.val.pointNotifyFunction.
This one gets called every time a calibration point succeeded or was
aborted. So that would allow you to play sounds after each successful point
collected instead of before. That means you don't have a sound for the
first point, but have a sound right before all the other points. Might work
for you?

Let me know what specifically you're stuck on if you need more info.

Cheers,
Dee

GJeanChar, 2020-02-18:
Hi Dee,
Sorry for the late reply, i was bedridden for the last week!
Thanks for the advice for the points presentation, it works beautifully!

Playing with the functions settings.cal.pointNotifyFunction and settings.val.pointNotifyFunction would not work for me, as I need the sound to be played while the point is presented for all the points, it's kind of an attention grabber to help children to focus on the point, i will look into the draw and new as you suggest!

I have another question though, i saw when looking into the closed issue, that you implemented a callback to replay only the non-valid points in the calibration, how can I use this?

Thanks a lot for all the help!
Cheers.
G.

dcnieho, 2020-02-18:
Ok, if you need a sound per point, those functions wouldn't work indeed.

`I have another question though, i saw when looking into the closed issue,
that you implemented a callback to replay only the non-valid points in the
calibration, how can I use this?`
What this does is not replay to replay non-valid calibration points, but
simply to "restart" the current point. I.e. a "new" will be sent to the
calibration function with the same coordinate as the current point. With
default key bindings, you activate this for the current point by pressing
the backspace key. So you could use this if you see the kid is not paying
attention, and it could cause the sound to be replayed.

I guess you already noticed, but you probably also want to set
settings.cal.autoPace
to 0, so that calibration (and validation) of each point needs to be
manually started by the experimenter pressing (by default) spacebar, giving
you full control together with backspace over the calibration procedure.

Cheers,
Dee

iandol, 2020-02-28:
Hi @GJeanChar -- this is somewhat off-topic, but I calibrate monkeys and so have similar problems to you with infants. I adapted Titta's callback to play avi movies (in my case a dancing monkey) and also make a sound when first displayed. The code is here: https://github.com/iandol/opticka/blob/master/communication/tittaCalMovieStimulus.m -- it uses my movieStimulus class, but it would be possible to factor out the drawing bits. I faced a couple of issues developing it: first PTB's movie playback doesn't play nicely with normal Screen flips, so you have to double buffer the movie frames; second: I wanted transparent movies, so had to work out what format is needed. The movies do grab attention well...

closed 2020-03-10 by dcnieho