dcnieho/Titta #15: Shift-escape and two screen calibration

iandol, 2020-03-04:
Hi Dee,

Shift-escape raises an error, and if you are using an operator and presenter screen, leaves both PTB windows open, so you cannot easily return to matlab. Although I normally ensure my focus is in the matlab command window and I can blind type 'sca' to get back, I wonder whether Titta running 'sca' or at least closing the operator window, and possibly running some other cleanup, like  ListenChar(0) etc.

https://github.com/dcnieho/Titta/blob/master/Titta.m#L803 is the place I think.

Something like: 

```matlab
case -5
    % full stop
    obj.buffer.leaveCalibrationMode();
    ListenChar(0); ShowCursor; Priority(0);
    sca
    error('Titta: run ended from calibration routine')
```

iandol, 2020-03-04:
Oh, just saw the Titta paper is published, congrats! :+1:

dcnieho, 2020-03-04:
Thanks for the congrats!
Ah, great idea. I always have a try-catch around PTB code, closing screen
etc upon error (including the demos), so never thought of this. But it
makes a lot of sense. Will do (next week).

By the way, ListenChar is not something i use on Windows, not needed. It is
something that should normally be used on Linux, right? Could you suggest
where in one of the demo files I could add it (then I'll do them all).
Thanks!

Regarding the latency peaks for online data, I have heard a new SDK version
fixing the problem will be coming in a few weeks.

iandol, 2020-03-04:
Yeah, I also normally have a `try` `catch`, but in a small test setup the calibration ended up outside my `try` which is how i found this.

I normally add ListenChar just before keyboard is polled, and then reset just after, so e.g. https://github.com/dcnieho/Titta/blob/master/demos/readmeMinimal.m#L54 

```matlab
ListenChar(-1);
EThndl.calibrate(wpnt);
ListenChar(0);
```

I didn't know Windows doesn't leak keypresses like Linux and macOS; interesting!

Yeah, they also told me a new SDK should fix this. For a bug which compromises core functionality, I am not impressed by how slowly they responded to this.

iandol, 2020-03-04:
I just tested KbCheck() on Windows 10 + MATLAB 2019b and I see the same behaviour, in that keys you press leak into e.g. the MATLAB editor if it has focus. Test code:

```matlab
KbName('UnifyKeyNames');
loop = 0;
while true
	[keyIsDown, ~, keyCode] = KbCheck(-1);
	if keyIsDown == 1
		rchar = KbName(keyCode);
		switch lower(rchar)
			case {'q','0'}
				break;
			case {'1','1!','kp_end'}
				loop = loop + 1;
				fprintf('Hello %i!\n',loop)
		end
	end
	WaitSecs(0.01);
end
```

If I run that from the editor, everytime I press 1 I get the print statement but also a 1 leaking into the editor. Maybe you don't use KbCheck or set up in a different way?

dcnieho, 2020-03-04:
Hmm, I don't get stuff in the editor when a PTB window is open (never run
without). It might actually be that keys do leak through, but never end up
in my code because running causes focus to shift to the command window on
Windows. It could all be over a decade of being used to clearing input from
the command window, though i think i remember not seeing any (not near a
PTB setup now).

But thanks for the tips on how one would use ListenChar, I'll implement
that in the demos.

I too wish Tobii would be faster about the SDK (and other issues: did you
notice the frequent 1-2 sample gaps during saccades in the Spectrum data,
or do your monkeys not show that?).

iandol, 2020-03-04:
Actually we are having serious problems getting past the calibration stage, we have yet to even collect data, I'll contact you offline for some advice...

dcnieho, 2020-03-04:
Sorry to hear that. I have little monkey experience but would be happy to
try and help, so do feel free to mail indeed!

closed 2020-03-10 by dcnieho