dcnieho/Titta #17: [Feature Request] Selectively recollect calibration points

iandol, 2020-03-07:
Babies or animals are not easy to calibrate, and often lose attention momentarily or randomly look too briefly at a target. Currently, we can "redo" a point **if** we are still in that point with <kbd>backspace</kbd>. However it isn't always easy to trigger this at the right time, and often you end up with a mix of good points and bad ones, which causes a calibration to fail. It would be really nice to be able to recollect points selectively. Tobii Studio allowed for this in its infant calibration mode  (however Pro Lab or ETM's calibration mode does not, among other things). It would be hugely beneficial to able to select calibration points. Eyelink uses a fairly simple strategy:

1. Go into a manual calibration mode.
2. Press <kbd>1</kbd>-<kbd>9</kbd> to trigger collection of that point.
3. <kbd>Space</kbd> to collect data, but your responsibility to press the number for next calibration point.

I had a quick look at DoCalPointDisplay() which is where I assume the logic for this would have to go. 

It would be great on an operator screen that the status of calibration points that were collected could be highlighted, or a text line with this info drawn. I realise this information may not be available however, calibrationCollectData() doesn't seem to return any status (possibly because the C SDK returns a status that only means no error?).

iandol, 2020-03-07:
Rough flow diagram of the fully manual calibration loop
[image]

dcnieho, 2020-03-10:
@iandol, interesting request! Would be willing to look into that. Lets see (read the whole thing first, this is partially a brain dump):

Ok, to clarify:
- only makes sense on two-screen setup, right?
- Would need to somehow mark each point if not yet collected or unsuccessful collection (`calibrationCollectData` returns success/failed for the specific point). 
- when no point selected, calibration display for participant should not show anything (guess i'll add an extra state 'empty' or 'nothing' besides 'new' and 'draw' and such to the calibration drawing function)
- Note that i have never seen Tobii Studio, so do not know what that looked like or what their mechanism was

Points to think about:
- How to deal with validation? at some point we need to switch mode and validate instead of calibrate. Note that in principle the API allows to fluidly switch back and forth, we can compute calibration after each collected point and collect validation data when we want, and then calibrate some more.
- Related: want the option to recalibrate specific points after validation? IIRC, `calibrationRetrieveResult` only returns failure or success, doesn't tell you if any specific points made it fail.

Design that i am thinking about:
- When you have an operator screen, there is a button there to switch to manual mode. Alternatively, in settings, you can request manual mode directly. There is no option to go back to automatic (we can use screen real estate better).
- When in manual mode, the array of calibration targets defined in settings is shown on the screen, along with eye images. So far, this is like the operator screen we have now.
- head display to participant is optional (don't think it is right now, bet you don't want to show it to them?)
- Each dot has a number for keyboard activation (if nine or less) and is clickable.
- This data collection either succeeds or fails. If success, see below. If fail, notify operator. I envision a status message in some corner saying 'collecting','failed','success' denoting status of data collection for individual calibration point.
- unless a point is currently collecting, nothing is shown on participant screen
- After each collected data point, we call the compute calibration function. That might fail or succeed, status of last attempt is noted somewhere on screen. Note that for 1 point it may succeed, but when adding more points it may start failing. This doesn't mean something is wrong with the newly added point(s), just somehow that all the collected points together fail some criterion (the complete lack of idea what failed in the calibration is a frustration i have registered with Tobii...). But at least you know what the state of your current calibration is.
- You can enter a delete mode, where the collected calibration data for a point you click (or number you press on keyboard) is discarded. After discard, calibration is also recomputed.
- When successful calibration computation, show the data collected for each point on the operator screen (thats those line displays, but for the calibration data). Point not yet collected/failed are encircled in red, currently collecting point has blinking green circle around it.
- At any point in time you can switch to validation mode. Operator screen then contains a different array of points, the validation points defined in settings. Can click/keyboard press like before, collected validation data shown like current display (with overlay and data quality stats).
- At any point in time can go back to calibration mode, where you can then delete or recollect specific points. When you changed anything about the calibration, the validation mode screen is a clean slate again (different calibration requires a new validation).
- Need to think about how to enable menu selecting among multiple calibrations. Do you envision ever needing that? E.g. after validation you are not happy, go back to cal, change some points, validate again, regret and wish to go back to the first?

Note that this way you can define way more points for calibration and validation in the settings than you intend to use, and then select during calibration which you'll actually use. This allows you to e.g. collect for a smaller grid if points in the corners didn't work.

Describing the practical issues, practical tricks and interface design for such a manual calibration procedure may well make for an interesting paper. There is also a trade off between complexity of interface and training of assistants. I can think of a colleague working with infants who may well be interested in this as well. Email me offline if you want to pursue that further (if you see a need/use for such things in your field), but lets keep discussion of the interface in the thread here.

iandol, 2020-03-10:
Thanks for a thorough reply! ;-)

> only makes sense on two-screen setup, right?

Actually, with the Eyelink, although it always has an operator mode (dedicated PC with realtime OS runs the show), there is no real interface for manual calibration, you select points without almost any feedback (forgetting if top-left is 1 or 5 etc), and it is a klunky affair. So even having a text log in the command window of which points were selected to test and which completed would be a hell of a lot better than the Eyelink. I suspect it would be easier for you to do. However online visual feedback really improves the experience tremendously, so two-screens would definitely be preferred.

> Would need to somehow mark each point if not yet collected or unsuccessful collection

Yes, an important part of getting manual calibration to work, although as you said, it may be that even 3 points in a 5 point calibration would give a succesful calibration. I'm normally used to thinking that if it is 5 point calibration, all 5 must be completed (as is the case for eyelink).

> when no point selected, calibration display for participant should not show anything

Yes, sometimes we need to take a small time out and give a bit of apple or something, so stuff on the screen would be confusing.

> Note that i have never seen Tobii Studio, so do not know what that looked like or what their mechanism was

As we are having problems with the firmware with our Spectrum Pros (why we can't calibrate at the moment so it turns out), we just started testing the TX300 and Tobii studio today, I'll be able to give you some more feedback on how infant mode is controlled tomorrow.

> Related: want the option to recalibrate specific points after validation?

Without us knowing which points are causing a poor validation I suppose it is hard to know which points, but I imagine there must be a spatial relationship with validation errors and their corresponding calibration points, so this may be a nice option to have.

> There is no option to go back to automatic (we can use screen real estate better).

Sounds reasonable to me.

> bet you don't want to show it to them

Correct!

> After each collected data point, we call the compute calibration function

I hadn't thought of that but that is cool if it is possible and doesn't interrupt calibration

> You can enter a delete mode

It could be like shift-click or shift-number for discard+recollect.

> Need to think about how to enable menu selecting among multiple calibrations

Because you can switch iteratively between cal and val quite easily, not sure how we define a previous calibration?

I'm happy to help where needed on this, and I think it will really benefit those of us working with non-verbal subjects.

iandol, 2020-03-11:
FYI from the Tobii Studio manual (sorry for poor formatting from PDF):

> Infant calibration is used when conducting studies on infants.Small animations(in AVI format)are displayed where the dots nor-mally appear.This worksas an attention grabber suitablefor smallchildren.It canalsobe usedwhenit is difficultfor someoneto followthe calibrationinstructions.TobiiStudiocanalsodisplaydifferentAVIsto attract the test participant’sattentionif theybeginto respondpoorlyor growtiredof the calibrationAVI.Theseitemsare calledattentionanimationAVIs.No calibrationdatais collectedduringthe displayoftheseAVIs.Codecsthatcanbe usedto makethe AVI filesare MicrosoftVideo1, Techsmith1or Xvid

> Infant calibratio nis an interactive calibration mode wherean operatorusesthe keyboardto controlthe calibrationprocedure.To showthe calibrationanimationselectedin the GlobalSettingsdialogbox,pressthe RIGHTARROW key duringthe calibra-tion.Whenthe participantis watchingthe animation,pressthe RIGHTARROW key againto startcollectingcalibrationdata.Itis assumedthatthe participantwill focuson the centerof the calibrationanimation.Whenenoughdatapointsare collected,thecalibrationwill moveon to the nextcalibrationpoint.Theoperatorcanagainpressthe RIGHTARROWkey to startcollectingdata.Thisis repeateduntilall calibrationpointshavebeendisplayed.Thefollowingkeysare usedto controlthe Infantcalibrationprocedure:•RIGHTARROW:Displaysthe calibrationAVI at the nextcalibrationpoint.If the calibrationAVI is alreadyvisible,a cali-brationis madeat thatpointbeforeproceedingto the next.•LEFTARROW:Pause.Hidesthe calibrationAVI or the full-screenattention-grabbingAVI•UP ARROW:Displaysthe attentiongrabbingAV

dcnieho, 2020-03-11:
Thanks for the discussion and info about Tobii Studio.

I think a combination of what Studio does and what the eyelink does, with a bit of own special sauce, will make for a very nice interface. There would be visual feedback all the time (on operator screen) about what is happening, and recalibration is as simple as selecting an already collected point.

Yeah, the call to compute the calibrate is really cheap on most of the eye trackers i have tested with (exceptions that i remember are the 4C and X2-60, but so be it for users of those, I consider it pretty niche that those systems would be used with difficult populations--though i don't have stats). Its necessary to call the calibration function to get that line display like we have on the validation result screen and also in the Studio screenshot.

Regarding number of calibration points, i guess that for this advanced use case, it would be nice to offer the possibility to skip a point that can't be calibrated and still continue. Practice will show how robust the eye trackers are to that.

I'm not sure how to define a "calibration" indeed. Either any state that precedes some collection of validation data, or perhaps have an explicit "save" button or so. Probably the latter is better, more flexibility. Downside might be that some think it must be pressed, but there would be no cost to that except the extra operator action.

Regarding delete, I think it should be distinct from recollect. So two things can happen on an already collected point:
1. recollect: normal click
2. delete: indeed like shift click or so.
Its perhaps a bit niche, but you may find that a specific point, even though some data was collected, is giving you trouble, so you want to just kick it out again. Not sure that ever happens, but why not offer the flexibility, as its super cheap to do.

Thanks for the input! I think perhaps the next step is for me to play with this, in a separate branch, so we have some mock up or so that you can test as well and see what you think.

iandol, 2020-03-12:
> Regarding number of calibration points, i guess that for this advanced use case, it would be nice to offer the possibility to skip a point that can't be calibrated and still continue. Practice will show how robust the eye trackers are to that ... even though some data was collected, is giving you trouble, so you want to just kick it out again...

Actually, single calibration points not calibrating is a pretty common experience for me at least, even sometimes for humans wearing glasses. 

I think having an explicit "save" allows an operator to "tag" a state, it would enable them to have more control, and you could call it something like "snapshot" to identify it as optional?

dcnieho, 2020-03-13:
Good idea to call it a snapshot! Ok, this has taken enough form now that I'll try to implement a first attempt.

@iandol, please do update to latest version of Titta now as well. I found that there was some logic error when you have multiple calibrations. In some situations, the validation result screen may not show the one currently selected. Also, in some cases Titta.calibrate() does not log the applied calibration in a message, even though one was applied. These are now fixed. Titta also gained a new option on the setup screen (with the head), press 'c' to change from which eye you are recording (if your eye tracker supports that). Its a by-default hidden button on that interface, but the accelerator key always works. Implemented this for another type of difficult subject, e.g. with squint.

iandol, 2020-03-14:
Ha, indeed one of our monkeys has strabismic squint, so this is potentially useful when we get our Spectrum Pro back working... Will have a test, thanks Dee.

dcnieho, 2020-05-08:
Hey! Its by far not done yet, but give the dev branch a try if you are curious. Don't use the snapshot menu button, it'll crash. Deleting point data is also not possible. Also, i haven't tested on Linux, but don't see why it wouldn't work :P Run `demos/readmeManualCalibration.m`

What you can do:
- the head display, you can move and resize it (hold shift to maintain aspect ratio)
- show points for calibration by clicking them, or pressing their number on the keyboard
- if you press multiple, they get enqueued to be shown in the pressed order. escape cancels all points for which calibration data collection hasn't started yet.

I don't have any specific things i'm looking for feedback on yet, but do tell me of course about thoughts that come up. Ah yeah: a known bug is that point collection status is not reset when changing to monocular calibration mode, or back. Just not implemented yet (shit there are many interacting parts to think about here!)

Oh, i see i missed a question. Indeed i think the TX300 doesn't do monocular calibration. But easy enough to check, if it does, the eye selection menu would be available in Titta, if it doesn't that menu isn't available.

iandol, 2020-05-11:
Hi Dee, I had a quick play just now, but even though I changed the presenter/operator screens (in linux/mac these are 0 and 1, not 1 and 2 as on Windows), somewhere in the drawing code, most of the operator display UI ended up in the presentation display anyway. I can step debug my way through but don't have the time to test more until around Wednesday...

dcnieho, 2020-05-11:
Hi Ian,

Thanks for giving it a try! Are you sure you did not swap around the
participant and operator screens? I just looked through the code, and for
instance the buttons should show up on the operator screen, i don't see how
they could end up somewhere else. I would really appreciate it if you could
look at why things end up on the wrong screen for you on Linux, when you
have time.

iandol, 2020-05-12:
Yes indeed, operator error ;-) I just defaulted to operator screen is going to be the first display and presentation the highest value returned by `Screen('Screens')`, but you do it the other way round, and I didn't check so got the numbers the wrong way round.

This is great!!! You've got most of this working the way I envisaged already, and being able to see as we collect points what the overall status is and recollect points is great. Some questions:

1) If a point is green but doesn't have the error lines, does that mean we tried to collect that point but no calibration data was valid? If so perhaps have an intermediate colour to more quickly identify these point.
2) Some points I'd collected went back to grey after collecting other points, perhaps they were points that were collected but didn't have valid data, they reset back to grey?
3) I do really like how you've managed to combine the two interfaces on the operator screen, the floating resizeable setup panel is super cool!!! Lowish priority, but we normally need to see the head position setup on the presenter as well as operator screen (this is currently an option in Titta master, usually if we have one operator and we have separate rooms for subject/operator they have to move our chair around and seeing it on the screen in the room where the subject is helps). I realise one problem is that the setup panel could obstruct calibration points if the operator forgets to toggle it off, and this may cause confusion for the operator. I'm not sure the best way round this (remind operator to toggle setup-panel off when collecting data points somehow, or always show setup-panel on operator screen and toggle it on/off only on presenter screen).

Let me know what you would like me to focus on if I can help with targeted testing. But this interface is already way better than anything I've used with the Eyelink or Tobii Pro Studio/Lab 👍 😀

dcnieho, 2020-05-12:
Hi Ian,

Great it works and that it's making you happy already! Let me answer your
points in alternative order:
2. This surprised me at first two, but some orders of calling lead to
different results: Doing first only the top left point leads to calibration
data collected (point green), but calibration failed (see top left of
screen). Then collecting the point in the middle leads to calibration
success, but that computation removes the top left point apparently. I can
see that by the calibration feedback i get from the Tobii, so i made that
point grey again. Doing first middle, then top left, both are used for the
calibration. Doing top-left and middle in one sequence (press them quickly
together, so second gets queued and shown directly after first, also leads
to both points being used for calibration. Nothing i can do about that i
guess, except communicate it the way i do now (let me know if you think of
a better way!) and operators will have to learn these dependencies i guess.
Note that I don't know yet if the spectrum always throws out data once
calibration fails, or only in the case described above. Its hard to get it
to fail after the center has been collected :p
1. Is this the same point as the previous question? Or perhaps not: this
may happen if you collect data for a point (e.g. top-left only) and
calibration fails. Then there is data for the point, but no lines to show
as they are generated by the calibration computation. I assume data for a
point is collected (if the point collection function succeeds, else they
turn red) until i hear otherwise from the calibration function (see above).
If you are using a spectrum again, can you describe a specific sequence
where you see this? Then i can see if i can reproduce if it is not expected
(NB, for as far as i know what to expect :P).
3. If you hold down shift while pressing the head button or keyboard key,
you should get the head display on both screens. It disappears on the
subject screen when a calibration point is shown and reappears after (not
sure why, i didn't implement that, might only be with the animated drawer,
need to look further into that). Does that do what you need? If operator
wants to see the screen but no longer to participant, toggle it off and
back on without holding shift down.

iandol, 2020-05-13:
Yes question 1 and 2 were basically the same question, and great that you you can use shift to toggle, that sounds fine (can't test today)...

dcnieho, 2020-05-23:
hey man, its basically done. Check out the dev branch.
Only the functionality to load previous calibrations when calling the
calibrate function is not implemented. Snapshot work. and it turns out
they're really snapshots: when you load one, you are back in the state when
you saved it, and you can add points to it. So you can use them to store
something you are happy with before you continue and try to improve, and if
your attempt to improve backfires, nothing is lost.
so shift-click on points to discard their data
three things to note:
1. when a calibration fails, the previous one stays active. Hence the
status message in top left now contains info about which points a currently
active calibration is based on. So if you see "calibration failed" this
means the last calibration attempt failed, not that you are not calibrated
at all. you are not calibrated at all only when there are no points listed
2. if you calibrate piecemeal, tobii recommends to start with either a
center point, or two points at opposite sides of the screen. I have also
noted something about the following order (for my eyes). if i first do
point 3 (middle), let it calibrate, and then queue up points 1 and 5
(top-left, bottom-right) immediately, sometimes this yields a successful
second cal, sometimes not. if not, recollecting 3 usually fixes it. Anyway,
it seems there is a lot of getting to know your machine here before you
know how to best approach such a flexible calibration for your tracker
3. snapshots only store a calibration state. You can later add validation
for that calibration, and the snapshot menu will be updated to reflect
that. Note also that if the currently active calibration corresponds to an
item in the snapshot menu, that item is highlighted. If not, no item is
highlighted.

I've reasonably tested this on Windows, and expect it to work on Linux too.
I hope you can seriously put it through its paces, see if you can get it to
crash or act unexpectedly. Do send me the callstacks for any crashes and
ideally also how to reproduce them (you can send these by mail, they may be
long). I'm of course also very interested to hear about how this actually
works for you.

dcnieho, 2020-05-25:
If you at some point test with a monkey, would you be able to make a screen
recording of the process? I would be very interested to see how you
actually use it.

iandol, 2020-05-26:
Yes, I'm happy to show you it in action with some subjects via mail.

I get a crash when using the snapshot menu sometimes, callstack is:

```
Unrecognized function or variable 'iValid'.

Error in Titta/doManualCalib (line 3920)
                        nElem           = length(iValid);

Error in Titta/calibrateManual (line 1065)
            out = obj.doManualCalib(wpnt,out,kCal,currentSelection);

Error in readmeManualCalibration (line 124)
    tobii.calVal{1} = EThndl.calibrateManual([wpntP wpntO]);
```

I'll work round it and continue testing.

dcnieho, 2020-05-26:
Thanks! Did you check out the latest dev branch HEAD? That code doesn't
appear on that line anymore, it looks like its about a week old or so.

iandol, 2020-05-26:
I'm on this commit according to Git:
```
Titta on  dev 
[I] ➜ git log -1
commit 347b0258e1d11a4e00ce136665e32cb1fe66f42c (HEAD -> dev, upstream/dev)
Author: Diederick Niehorster <dcnieho@gmail.com>
Date:   Mon May 25 13:18:39 2020 +0200

    snapshots didn't work when there were discarded points
```
But something seems to have gone wrong with my local repo, so I recloned and it seems to be back in sync. Don't see that error now and indeed snapshot menu is working great!!! Workflow is really great. There do indeed seem to be some strange order of collection issue, I'll try and see if I see a similar pattern to you. We'll probably give this a quick test with subjects tomorrow and I'll give you some more feedback then.

It's really nice, thank you for all your hard work on making this interface a reality!

dcnieho, 2020-05-26:
good to hear and thanks! Yeah, we're just working with what the eye tracker
does, the interface does its best to report what the actual state is, no
matter how you got there. I guess it'll be important for your operators to
have experience with the system and know what they should and shouldn't do
to not lose points (since points are precious). I'm of course happy to hear
about the things you learn and see if i can reproduce them! And thanks for
the thanks! It was quite fun and hope it'll be useful for you and more labs

iandol, 2020-05-26:
Here are a few observations:

1) [shift]+[h] doesn't seem to work anymore to show head on both displays, although [shift]+click button does.
2) calibration.selectedCal used to return a single number, but now it seems to return something like [1 X], even if there is only one cal in the attempt array:
```
           type: 'manual'
    selectedCal: [1 5]
     wasSkipped: 0
          flips: [1×4206 double]
        attempt: {[1×1 struct]}
         status: 1
```
3) Trying to run `getValidationQualityMessage(calibration)` tells me:
```
'Reference to non-existent field 'status'.
     
     Error in Titta>@(x)x.status (line 1826)
                     iVal    = find(cellfun(@(x) x.status, cal.val)==1,1,'last');
     
     Error in Titta.getValidationQualityMessage (line 1826)
                     iVal    = find(cellfun(@(x) x.status, cal.val)==1,1,'last');
```
4) Are you planning to add back the ability to pass a previous calibration or is that now not possible for manual mode?
5) It rocks! ;-)

dcnieho, 2020-05-26:
>    1. Trying to run getValidationQualityMessage(calibration) tells me:
>    1. Are you planning to add back the ability to pass a previous
>    2. It rocks! ;-)

Thanks!
1. thats a regression, will fix
2. yes, for manual calibration, it returns which attempt, and within that
attempt, which cal (note there is a cell-array 'cal' in each attempt
3. Ah, you can't pass one of these calibrations directly to that function.
I agree that that should be possible. will fix
4. yes, thats the one todo left.
5. great!

dcnieho, 2020-05-26:
hi Ian,

Don't pull any new versions from the dev branch anymore, just continue
testing with what you have now (unless the inability to show the head on
the participant screen is blocking you--let me know i that is the case). I
am pushing some untested things, so the HEAD may be broken.

iandol, 2020-05-27:
Yeah I just saw the changes, I can just detach my HEAD to the last safe commit. By the way, I did test your `*untested*: implemented that getValidationQualityMessage works on the output of calibrateManual()` but it errors out at the moment (untested as it says on the tin ;-), so I just detach to the previous to that commit.

dcnieho, 2020-05-27:
good, that should probably work. The commit before is also unteste, but
very low risk. It should bring back holding down shift to show head (and
online gaze) on both screens.

iandol, 2020-05-27:
I can confirm [shift]+[h] is working again :+1:

dcnieho, 2020-05-27:
nice! dev HEAD should work again now. I made some major naming changes in the mex files, so its probably best to do a clean clone. TODO still is loading previous calibrations

iandol, 2020-05-28:
OK, `getValidationQualityMessage` now works. I also had to rebuild the MATLAB path (as you renamed the folders). You maybe can add a small function like this to your Titta root folder:

https://github.com/iandol/opticka/blob/master/addoptickapaths.m

To automatically regenerate the correct path (removing .git and any other path locations specified via `pathExceptions` that MATLAB does not need). A bit easier for general users.

iandol, 2020-05-28:
A few small points on the settings:

* Why `settings.mancal.paceDuration`, rather than `settings.mancal.cal.paceDuration` as you also have a `settings.mancal.val.paceDuration` and it is more similar to `settings.cal.paceDuration`?
* What is the difference between `settings.mancal.bgColor ` and `settings.UI.mancal.bgColor` and for other similar settings?

dcnieho, 2020-05-30:
thanks for the comments on the settings, will check. I think i found
`mancal.cal` a bit redundant, and thus left that out. Symmetry and logical
grouping (proper deepnesting) is better here indeed, will change.

The various bgColors are for participant and operator screen, and
backgrounds of overlays, such as the head positioning overlay. I'll
document better.

Interesting point regarding adding to path. I always simply
addpath(genpath()) and be done with it. Matlab doesn't seem to have an
issue with .git subfolders on the path. I'll look into providing this
function though (thanks for sending me the example!) to provide a clean way
of adding Titta to the path for those who want.

dcnieho, 2020-06-01:
ok, i've done the settings related thing you asked about. will look at
paths next, but not today

iandol, 2020-06-02:
No worries, Here is the path script I use modified for Titta:

```matlab
mpath			= path;
mpath			= strsplit(mpath, pathsep);
opath			= fileparts(mfilename('fullpath'));
for i = 1:length(mpath)
	if ~isempty(regexpi(mpath{i},opath))
		rmpath(mpath{i}); % remove any old path values
	end
end
opaths			= genpath(opath); 
opaths			= strsplit(opaths,pathsep);
newpaths		= {};
pathExceptions = [filesep '\.git|' filesep 'src'];
for i=1:length(opaths)
	if isempty(regexpi(opaths{i},pathExceptions))
		newpaths{end+1}=opaths{i};
	end
end
newpaths = strjoin(newpaths,pathsep);
addpath(newpaths); savepath;
disp('--->>> Added Titta to the path...')
clear mpath opath opaths newpaths pathExceptions i
```

dcnieho, 2020-07-02:
@iandol, I have now implemented the reloading of previous calibrations, please give it a try. When reloading, snapshots and validations are lost, just the cal is kept.

Could you put all of this through its paces thoroughly? And if you could take a video of the process showing how you'd use it, please do and send it to me.

This is all now on the master branch.

By the way, the 1.8 SDK is out, but the Linux link still refers to the old 1.7 SDK. I'll update when i can.

iandol, 2020-07-03:
Hi Dee, great thank you! We're not doing any more experiments till next week, but I will test this a bit on my spectrum (N=1).

I will try and record something for you next week.

Do you know what the 1.8 update does (didn't see a changelog in the downloads)?

iandol, 2021-04-22:
I think we can call this closed ;-)

closed 2021-04-22 by iandol

dcnieho, 2021-04-23:
haha :) Working well for you or are there things that have come up now that
you've been using it for a while? No guarantee i'll do something about it,
but always interesting to hear about