dcnieho/Titta #19: Error in textbounds with ° symbol, probably encoding related?

iandol, 2020-03-10:
Hi Dee,

On Windows 10 and MATLAB 2019b we are getting a Windows specific error from DrawFormattedText2 whenever calibration results are displayed, called via Titta/showCalValResult -- I encountered this last year when first trying Titta and determined the problem was caused by the ° symbol, but then moved to Linux where this doesn't occur and I cannot currently connect to Gist where I store my notes... I assume this depends on the locale settings in Windows. I know this is not actually a Titta issue (though Titta always shows ° symbols) , and I will have a better look to find my notes when gist comes back online, but as you know windows better than I and you wrote DrawFormattedText2, I just thought I'd document it here.

```
Error in function TextBounds: 	Usage error
Invalid multibyte character sequence detected! Can't convert given char() string to Unicode for DrawText!

Error using Screen
Usage:

[normBoundsRect, offsetBoundsRect, textHeight, xAdvance] = Screen('TextBounds', windowPtr, text [,x] [,y]
[,yPositionIsBaseline] [,swapTextDirection]);

Error in DrawFormattedText2 (line 491)
            [~,bbox,h,xAdv] = Screen('TextBounds', win, subStrings{q},0,0,1,righttoleft);

Error in Titta/getTextCache (line 1808)
                [~,~,txtbounds,cache] = DrawFormattedText2(text,'win',wpnt,'cacheOnly',true,args{:});

Error in Titta/showCalValResult (line 2739)
                            valInfoTopTextCache = obj.getTextCache(wpnt(end),valText,OffsetRect([-5 0 5
                            10],obj.scrInfo.resolution{1}(1)/2,.02*obj.scrInfo.resolution{1}(2)),'vSpacing',obj.settings.UI.val.avg.text.vSpacing,'yalign','top','xlayout','left','baseColor',obj.settings.UI.val.avg.text.color);

Error in Titta/calibrate (line 818)
                [out.attempt,kCal] = obj.showCalValResult(wpnt,out.attempt,kCal);

Error in tobiiManager/trackerSetup (line 303)
					me.calibration = me.tobii.calibrate([me.screen.win me.operatorScreen.win]); %start calibration
```

iandol, 2020-03-10:
Here is a minimal testcase, I tried with using ALT+0176 to insert it in MATLAB editor or using `char(176)` as you do in Titta, same result. I can get it to work, if I edit it in a unicode editor, inserting the correct unicode character there, although it reders badly in the MATLAB editor, PTB displays it just fine...:

```matlab
unction texttest()

bgColour = 0.5;
screen = max(Screen('Screens'));
screenSize = [];

ptb = openScreen(screen,bgColour,screenSize);

text1 = 'Here is working text...';
%text2 = ['Here it is 22' char(176) 'to the max'];

try
    vbl(1)=Screen('Flip', ptb.win);
    while vbl(end) < vbl(1) + 5
        DrawFormattedText2(text1,'win',ptb.win,'sx',100,'sy',100);
        DrawFormattedText2(text2,'win',ptb.win,'sx',100,'sy',200);
        vbl(end+1) = Screen('Flip', ptb.win, vbl(end) + ptb.ifi/2);
    end
    Screen('Flip', ptb.win);
catch ME
    getReport(ME)
    sca;
end

end

%----------------------
function ptb = openScreen(screen, bgColour, ws)
ptb.cleanup = onCleanup(@myCleanup);
PsychDefaultSetup(2);
Screen('Preference', 'SkipSyncTests', 2);
if isempty(screen); screen = max(Screen('Screens')); end
ptb.ScreenID = screen;
PsychImaging('PrepareConfiguration');
PsychImaging('AddTask', 'General', 'FloatingPoint32BitIfPossible');
[ptb.win, ptb.winRect] = PsychImaging('OpenWindow', ptb.ScreenID, bgColour, ws, [], [], [], 1);
[ptb.w, ptb.h] = RectSize(ptb.winRect);
screenWidth = 405; % mm
viewDistance = 573; % mm
ptb.ppd = ptb.w/2/atand(screenWidth/2/viewDistance);
ptb.ifi = Screen('GetFlipInterval', ptb.win);
ptb.fps = 1 / ptb.ifi;
end

%----------------------
function myCleanup()
disp('Clearing up...')
sca
end
```

I have confirmed that my user and system locale are both set to US English.

dcnieho, 2020-03-10:
@iandol, thanks for bringing this to my attention! There are frequent encoding issues with characters exceeding ASCII 127, the degree symbol being one of them. I have seen it break reliably when using on computers with Chinese code page, and on Macs. Hmm. char(176) was supposed to fix this, but apparently not. When you have your working string with unicode character, if you do `double('the srting'), what value is the working degree character? I am asking as i just looked up the unicode value for the degree symbol, but its also 176. Also, does this work for you:

```
text2 = ['FAIL: 22' char(0) 'to the max'];
text2 = double(text2);
text2(text2==0) = 176;
```
and if that works, how about this?
```
text2 = ['FAIL: 22' char(176) 'to the max'];
text2 = double(text2);
```

dcnieho, 2020-03-10:
And if the bottom suggestion works for you, could you try to add the following line to the top of the function `function [cache,txtbounds] = getTextCache(obj,wpnt,text,rect,varargin)` (currently on line 1805 in HEAD):
`text = double(text);`
Cheers!

iandol, 2020-03-10:
Thanks a lot Dee, I'll check when I get back in the lab in the morning!

iandol, 2020-03-11:
Yes,`double()` fixes the crash and the ° displays as it should; Titta runs fine when `text = double(text);` is added to `getTextCache()`...

Thanks Dee, gosh the sooner MATLAB on Windows support unicode the better :-(

dcnieho, 2020-03-11:
Thanks for the test, glad thats a viable fix. Yeah, i have run into this more often, also when sharing code that plots data with degrees on the axis...

dcnieho, 2020-04-13:
R2020a now does unicode, its in the release notes that new files are by default saved as UTF8. I'll give that a spin

iandol, 2020-04-14:
Cool, and finally!!!!!

dcnieho, 2020-04-14:
Let me close this again, I just realized that we need to stay compatible with users on older versions, who may run into the same issues as we did if they happen to open the file. We have a working solution, lets not touch it. Still, cool indeed for things that are less widely spread!

closed 2020-04-14 by dcnieho

iandol, 2020-04-15:
Yeah, the speed most people update MATLAB, we may be able to depend on MATLAB utf8 encoding by the year 2030 🙄

commit 9a307747 (2020-03-11): Fixing problems with encoding of degrees symbol, thanks @iandol!
Closes #19.