dcnieho/Titta #20: Deleted symlinks

iandol, 2020-03-14:
Hi Dee, you deleted the symlinks to the tobii shared library

https://github.com/dcnieho/Titta/commit/a1683325853732d0fe84d33dca4c3a4929fe0fcb#diff-c932750b1ab1eed5e61e81acf079213b

This causes the mex file to fail to load, I've readded the symlinks myself but wanted to mention this...

dcnieho, 2020-03-14:
> a168332#diff-c932750b1ab1eed5e61e81acf079213b

Oh! The symlinks don't work well on some of my windows boxes, apparently i
accidentally committed the trouble on one of them. Thanks for telling me,
will fix soon.

iandol, 2020-03-14:
When I installed git for windows it asked to install support for symlinks, so not sure if that is needed or you faced a different issue (Windows obviously doesn't support POSIX symlinks, and perhaps they get converted or something).

BTW, eye selection works great on our Spectrum, very nice indeed; the select calibration shows which eyes on each calibration so very easy to use!!! :+1:

dcnieho, 2020-03-14:
fixed with f6e55a28032017223e8c77980bf118feda556cf0. Closing

closed 2020-03-14 by dcnieho