dcnieho/Titta #21: [Feature Request] Loading previous calibrations?

iandol, 2020-03-14:
`calibrate()` returns a structure `CALIBRATIONATTEMPT`, and I was wondering whether it would be possible to allow that structure to be passed *into* a subsequent `calibrate()` call, in which case the (p)revious calibrations screen would come up first? That way we could save subject calibrations and call them up later.

dcnieho, 2020-03-14:
Certainly, good idea. Workflow would be simple, if passing one or more
previous attempts, they will be used to prepopulate the internal attempt
list. So then the "previous calibrations" button (or whatever the text
exactly is, not near a computer now) becomes available immediately, upon
which you can go to the validation result screen, select the one you want,
and revalidate (strongly recommended!).

Would that work?

iandol, 2020-03-14:
Yes exactly! 😄

closed 2020-03-14 by dcnieho

dcnieho, 2020-03-14:
And glad to hear that the eye selection feature works nicely for you!

iandol, 2020-03-16:
Hi just to say this is working great, easy to pass in calibrations and access them, then revalidate from that calibration, make new calibrations and select among them, really nice!

iandol, 2020-03-16:
I see one bug, if you are in dummy mode, you get a crash as it is not updated to handle the function parameter input.

```
'Error using TittaDummyMode/calibrate
     Too many input arguments.
```

dcnieho, 2020-03-16:
Thanks, fixed! It seems my testing surface is getting a bit large, sorry
that you end up doing some of it. Glad it works well for you. Note that you
can manipulate the struct you pass in to only contain the calibration you
actually want to use (in case it took multiple attempts before). Make sure
selectedCal is changed too to point to the right attempt element.

iandol, 2020-03-16:
No worries. One thing I think perhaps also worth adding (thanks for the date, very useful) to the metadata of the returned structure is the tracker model that was used. End users can also add that easily, but it seems something that would be generally useful by default. I don't know how applying the binary data from one tracker would affect another model, but keeping this data and metadata together makes it easy to know why if things fail...

dcnieho, 2020-03-17:
Good idea. Added in 239319c6d4b7a72b2085d84d2c1dc56f322f9dbe

commit 042f7300 (2020-03-14): calibration updates:
- calibration attempts, successful calibrations and successful validations now have timestamps
- moved checking if calibration loaded successfully into the loadOtherCal() method to allow for reuse
- can now provide previous calibrations to Titta.calibrate() to enable loading calibrations done at another time. Thanks @iandol for the idea!
Closes #21