dcnieho/Titta #23: When starting calibration with a previous calibration file, button is mislabelled

iandol, 2020-04-15:
Hi dee, we are loving the reload previous calibration. One very minor issue is the green `[calibrate (spacebar)]` button actually runs a validation:
[image]

Running a validation is fine for use, so we don't mind the functionality, but my operator thought he was doing a 1 point calibration each time then got confused it didn't seem to create menu entries so perhaps this can be rephrased to `[revalidate (spacebar)]`?

dcnieho, 2020-04-15:
Hi Ian,

What happens if you press that calibrate button is that you actually start
a calibration, same as when you don't load in any previous calibration, or
when you arrive back at this setup screen from the validation result
screen. The way to only revalidate a previous calibration loaded in is to
not press calibrate at this screen, but previous calibrations, and to then
on press revalidate on the validation result screen that you get to.

Is this not what you are observing?

You can change the text on that button if you want by editing
settings.UI.button.setup.cal.string

Cheers,
Dee

iandol, 2020-04-16:
I have our setup as 3 calibration points `[0.2 0.5], [0.5 0.5], [0.8 0.5]`, and my validation is set up for '[0.5 0.5]'. When I do a clean calibration (no previous cal file), `[calibrate (spacebar)]` uses the `[0.2 0.5], [0.5 0.5], [0.8 0.5]` points and then the validation `[0.5 0.5]` (4 points in total). 

BUT when we load a previous calibration, the `[calibrate (spacebar)]` button just presents `[0.5 0.5]` (1 point in total) — that's why I assume it is validating not calibrating (and IIRC no new entry is made in the [other calibrations] menu.

dcnieho, 2020-04-16:
You are right. I did a shitty test yesterday, can reproduce now. Should be
fixed now

closed 2020-04-16 by dcnieho

dcnieho, 2020-04-16:
I added another commit that does the following: when loading a previous
calibration in call to Titta.calibrate(), setting settings.UI.startScreen =
0 now brings you straight to the validation result screen from which you
can revalidate. This means you can skip the setup screen and pressing on
"previous calibrations" there, if you want. NB, without a previous
calibration, settings.UI.startScreen = 0skips the setup screens and shows
the calibration dot display at once

iandol, 2020-04-17:
Hi, when there is a previous calibration (line 731) but `startScreen` is true (line 754), `qGoToValidationViewer` is not defined and there is an error on line 809

https://github.com/dcnieho/Titta/blob/master/Titta.m#L731
https://github.com/dcnieho/Titta/blob/master/Titta.m#L809

dcnieho, 2020-04-17:
Ugh! I am so sure i tested this, but obviously i haven't. Fixed, thanks for
the report!

commit c3d0bb8b (2020-04-16): calibrate: loading previous calibrations fixed up: calibrate button on calibrate screen now calibrates again instead of adds a validation to the loaded previous calibration. loaded previous calibration also do not any longer have their fields changed. Thanks @iandol! Closes #23
commit 62d12227 (2021-06-09): readmeTwoScreens: fix calls to drawLiveData.

Thanks @iandol.

Closes #23