dcnieho/Titta #26: [Question] Positioning data stream?

iandol, 2020-08-11:
We have problems where sometimes the face position is lost, with our subject still trying to do a task. This is an error of the Tobii, rather than the subject. At the moment we are only using the gaze stream, but valid/invalid cannot distinguish between an .e.g. blink vs the loss of the face resulting in loss of eye signal. Is the resolution of this stream the same as the gaze data, it doesn't get timestamped correct?

dcnieho, 2020-08-11:
For almost all Tobii eye trackers (except perhaps the new Nano and Fusion),
the positioning stream should give you the same information as the 3D
positioning info in the gaze samples. Its kinda like a scaled trackbox
position if i understand correctly. And indeed, that stream does not get
time stamped. Does it allow you to distinguish blink vs loss of face? One
way to know that (at 5 Hz) is to look at the eye image stream (for the
Spectrum at least), you get different resolution images when tracking face
vs when not

iandol, 2020-08-18:
Hi Dee,

Good idea, I'll have a look at the images alongside that stream, see if I can work out if there is a consistent change when the face gets lost...

closed 2020-08-18 by iandol