dcnieho/Titta #28: some problem with TalktoProLab

dujiahei, 2020-09-05:
Hi,
 I meet some problem when using readmeProLabIntegration.m. I use the psychtoolbox to present my stimulus, the recording and stimulus present in one or two PC all is ok. Now I try it in one PC.
 I open the external mode in Tobii Pro Lab, and run the readmeProLabIntegration.m, but failed. I have no idea why and how to solve it.
 Thank you! 
[image]
[image]

dcnieho, 2020-09-05:
Hi, thanks for your interest!
I can't read the error message, but i am guessing it says something like
WebSocketClient unknown or not found? Is the
folder \talkToProLab\MatlabWebSocket empty?
If so, please have a look at
https://github.com/dcnieho/Titta/#how-to-acquire to see how to make sure
the right stuff is in there. If thats not the problem, please translate the
error message for me.

dujiahei, 2020-09-07:
Thanks a lot, this problem is solved by adding the jar file to WebSocketClient folder.
And I want to know how to use Titta and Tobii Pro Lab with multi-computer setups? Such as, The PC-A run and present the stimulus, and PC-B acquire the data.

dcnieho, 2020-09-07:
Hi, glad to hear! I'll try and see if i can improve that error message so
that it becomes more clear what to do :)

Is it an absolute requirement for you to present the stimulus from PC-A,
but collect the data on PC-B? While it is technically possible,
TalkToProLab is currently only implemented for single computer setup.
Someone with good programming skills is needed to implement the
two-computer setup case. the issue is that PC-A needs to send timestamped
stimulus events to PC-B, and that the times it sends need to be in the time
of PC-B, not PC-A. So the clocks of the two computers need to be
synchronized and remain synchronized for this to work. That logic is not
implemented now.

dujiahei, 2020-09-15:
Hi, I have no idea about how the stimulus sent to Tobii Pro Lab.  
Is findMedia and uploadMedia function in TalkToProLab.m? The two function is enough?
I implement it on one PC. For the two-computer setup case, I think maybe need some hardware for synchronization.

dcnieho, 2020-09-15:
Hi,
Have a good look at the demo code, it shows there are two stages for the
pro lab integration:
1. upload stimuli to Pro Lab, using uploadMedia indeed. You only have to do
this once. (Note that the examples also show sending AOIs. You can skip
that and make them in Pro Lab instead if that is easier for you).
2. For each run of the experiment, inform Pro Lab when a specific stimulus
is being shown. you do this using sendStimulusEvent.

The problem we have here is that the "when" needs to be in the time of the
Pro Lab computer, not the experiment computer. There are software
synchronization protocols running over ethernet that would be sufficient to
figure this out, but they are not implemented. No special hardware required
though.

I understand that you are able to run everything on a single computer?
Good! that would be a better way to go i think, less fragile for sure!

dcnieho, 2020-09-22:
i have added a better error message for the original problem, closing this as i understand you have solved your problem by running on a single computer

closed 2020-09-22 by dcnieho