dcnieho/Titta #33: [QUESTION]: Starting and stopping recording, best practice?

iandol, 2021-08-02:
Hi Dee,

With the Eyelink, the usual practice is to start and stop recording on every trial. I was curious what your view is on doing the same (setting `tobii.buffer.start('...')` and `tobii.buffer.stop('...')` and friends) for each trial? Or do you think it is better just to start recording and only stop at experiment end (sending trial marker messages for segmenting the analysis of course)?

dcnieho, 2021-08-02:
Hi Ian,

I think it may be best to just keep it recording and later cut up your data
based on messages. This because starting tracking takes some time (couple
hundred ms? not sure ad might be variable as eyes need to be found) and
thus complicates stuff.

In other news, new spectrum firmware and SDK versions are imminent, which
are promised to further solve the online data latency issues we've seen.
The huge spikes of over 100ms should already be fixed in the 1.8 SDK that
is currently out, but i am told it should now come down to only rare spikes
below 10 ms or so. I'm curious to see what we get!

iandol, 2021-08-03:
OK, thanks for the advice! I want to create the same wrapper API for the eyelink and Tobii systems, so have to standardise their behaviour while there are many differences in their requirements...

A new SDK will be very welcome, as also currently the 1.8 SDK reliably crashes MATLAB on Ubuntu 21.04 which I have installed on my dev workstation due to other dependencies... And we are very much hoping for the Spectrum firmware to fix our issues. 👍🏾 👍🏾 👍🏾

dcnieho, 2021-08-03:
Do you have any insight into whether its my code or the sdk that causes the
crash? Even if the stack trace ends up with an sdk function, it may well be
me

iandol, 2021-08-03:
It isn't your code, I tested the SDK directly and got the same stacktrace crash occuring in libavahi...

dcnieho, 2021-08-03:
good to know (**phew**). You reported it and got a good response? I can
help send it directly to the right person if it would help

iandol, 2021-08-03:
I reported it on the Tobii support site, case number 00503093 — I was told the SDK does not support the latest Ubuntu generically, and the next version will. So I assume they closed the case without registering the potential issue, but will face the same crash when they do test it (full crash log in in that case number, excerpt below):

```matlab
Stack Trace (from fault):
[ 0] 0x00007fd420047fbb /lib/x86_64-linux-gnu/libc.so.6+00266171 gsignal+00000203
[ 1] 0x00007fd42002d864 /lib/x86_64-linux-gnu/libc.so.6+00157796 abort+00000278
[ 2] 0x00007fd42002d749 /lib/x86_64-linux-gnu/libc.so.6+00157513
[ 3] 0x00007fd42003f3d6 /lib/x86_64-linux-gnu/libc.so.6+00230358
[ 4] 0x00007fd193c5cd9f /lib/x86_64-linux-gnu/libavahi-client.so.3+00040351
[ 5] 0x00007fd13aed2a6a /home/cog5/Code/TobiiSDK/lib/64/tobiiresearch.mexa64+00633450
[ 6] 0x00007fd13aed2b73 /home/cog5/Code/TobiiSDK/lib/64/tobiiresearch.mexa64+00633715
[ 7] 0x00007fd13ae8ccfe /home/cog5/Code/TobiiSDK/lib/64/tobiiresearch.mexa64+00347390
[ 8] 0x00007fd13ae7d19c /home/cog5/Code/TobiiSDK/lib/64/tobiiresearch.mexa64+00283036
[ 9] 0x00007fd420430450 /lib/x86_64-linux-gnu/libpthread.so.0+00037968
[ 10] 0x00007fd42011ed53 /lib/x86_64-linux-gnu/libc.so.6+01146195 clone+00000067
```

dcnieho, 2021-08-03:
Ah great, guess it's on their radar then!

dcnieho, 2021-09-17:
@iandol: the new SDK (and spectrum firmware) is out, and i have just updated Titta. Support for new Ubuntu is in the release notes indeed (though i didn't update my Linux system). And indeed, with new SDK and firmware (issues were solved on both sides), sample latencies seem very good now

closed 2021-09-17 by dcnieho

dcnieho, 2021-09-17:
Very good -> at 600 Hz i get latencies (systemtimestamp at which i read sample from Titta's buffer - systemtimestamp of data sample) that are almost always very close to 3ms, with very rare spikes of up to 6 ms. That is, when track is established. When track is lost (like properly lost, e.g. looking away, to where the tracker goes into search mode), latencies hover around 8ms. Given that there is no data anyway, i guess that is fine.

iandol, 2021-09-17:
Great, I've tested V1.9 SDK and Titta and neither are crashing at least, although for some reason while calibration is working, in Titta I can't get the eye signal. Probably an issue with my wrapper code (as the calibration works), but I don't have time to debug, if I find anything that is not my mistake, I'll post it. 

Thanks so much for your fast updates keeping Titta at the cutting edge ☺️

dcnieho, 2021-09-17:
No worries! Always happy to hear if you find any problems. But indeed, if
you get validation data in the calibration interface, then data is coming
in from Titta's buffers.