dcnieho/Titta #34: Cannot set frequency

JunSapporo, 2021-11-10:
Hi,

I get this error since I updated Titta from last year version. I'm using this with ubuntu 20.04, matlab 2020a, 3.0.17 - Flavor: Debian package - psychtoolbox-3 (3.0.17.12.dfsg1-1~nd20.04+1). Did I have to set the frequency somewhere to run the demo?

thank you,

> >> smoothPursuitDemo
> --->>> Added Titta to the path...
> Error: Titta/init (line 586)
> Titta: Error setting tracker sampling frequency to 600. Possible tracking frequencies for this Tobii Pro Spectrum are [150,
> 120, 60].
> Raw error info:
> Error: TittaMex_
> Titta::cpp Error: Titta::cpp: Cannot set eye tracker frequency
> Error code: 209: TOBII_RESEARCH_STATUS_SE_INVALID_PARAMETER (An invalid parameter was given to the method)
> 
> 
> Error: TittaMex/cppmethod (line 76)
>             [varargout{1:nargout}] = this.mexClassWrapperFnc(methodName, this.instanceHandle, varargin{:});
> 
> Error: TittaMex/set.frequency (line 247)
>                 this.cppmethod('setFrequency',double(frequency));
> 
> Error: Titta/init (line 582)
>                 obj.buffer.frequency = obj.settings.freq;
> 
> Error: smoothPursuitDemo (line 79)
>     EThndl.init();
> 
> Error: smoothPursuitDemo (line 79)
>     EThndl.init();

dcnieho, 2021-11-10:
Thanks for your interest in Titta! It is strange that this error has not come up before, since 600Hz has always been my default for the Spectrum. It appears your Spectrum has a maximum sampling frequency of 150 Hz, is that correct? Is that also what you see in the Tobii Eye Tracker Manager?

Yes, you need explicitly set the sampling frequency you want to run the eye tracker at, if its different from the default. In the demo you are running, somewhere below line 40 you can write `settings.freq = 150;`

Let me know how you get along with it!

JunSapporo, 2021-11-15:
Thank you, it worked!

dcnieho, 2021-11-15:
Great!

closed 2021-11-15 by dcnieho