dcnieho/Titta #38: Linux: libtobii_research.so - file too short error

iandol, 2022-07-15:
Hi Dee,

With the latest version of Titta, we are seeing this error when initialising:

[image]

Is Titta working OK for you? Note I am physically stuck away from my workstations so can't access them directly, hard to debug at the moment...

dcnieho, 2022-07-15:
Hi Ian,

I am away as well, back a week from now. I have tested this on Linux, so
its surprising. Have you tried recloning, while matlab is not running (it
may keep some handle to the files, making them unwritable)?

Cheers,
Dee

iandol, 2022-07-15:
I think I found the problem, 

```
❯ ls -l
total 10960
-rw-r--r-- 1 cog5 cog5      27 Jun 30 03:38 libtobii_research.so
-rw-r--r-- 1 cog5 cog5      27 Jun 30 03:38 libtobii_research.so.1
-rw-r--r-- 1 cog5 cog5 5272272 Jun 30 03:38 libtobii_research.so.1.10.1
-rw-r--r-- 1 cog5 cog5 5633144 Jun 30 03:38 TittaMex_.mex
-rw-r--r-- 1 cog5 cog5  299632 Jun 30 03:38 TittaMex_.mexa64
```

For some reason the symbolic links `libtobii_research.so` and `libtobii_research.so.1` do not link to `libtobii_research.so.1.10.1` (they should show a -> arrow to identify their target) — it may need to wait the weekend to test though...

dcnieho, 2022-07-15:
Strange, i guess the symbolic links didn't survive the zip maybe? On the
github interface (all i can check now) they look like symbolic links.

xcwangbcp, 2022-07-23:
I am the staff who used the zip. I will try to use clone later with a vpn.

iandol, 2022-08-15:
Confirmed that fixing the symlinks fixed this problem so I'll close this...

closed 2022-08-15 by iandol

dcnieho, 2022-08-15:
Great. To be sure, this was not a problem with the symlinks in the Titta
repo?