dcnieho/Titta #39: Calibration fails if eye openness is not supported

eann, 2022-07-27:
Hi

In our lab we're using Titta with a Tobii Pro Fusion. I noticed that, in the latest version, when the toolbox tries to run the calibration routine it also tries to set the eye openness stream. This causes the calibration to fail if the eye tracker does not support it.

I made a quick hotfix that I can put in a pull request, if you'd like.

dcnieho, 2022-07-27:
Thanks for the report! This surprises me, the idea was for the stream to be
optionally requested. Hmm. Can you send the error message you get?
Thanks!

eann, 2022-07-28:
Sure! The error I get is

`
Error using TittaMex_
Titta::cpp::setIncludeEyeOpennessInGaze: Cannot request to record the eyeOpenness stream, this eye
tracker does not provide it

Error in TittaMex/cppmethod (line 76)
            [varargout{1:nargout}] = this.mexClassWrapperFnc(methodName, this.instanceHandle, varargin{:});

Error in TittaMex/setIncludeEyeOpennessInGaze (line 345)
            prevEyeOpennessState = this.cppmethod('setIncludeEyeOpennessInGaze',include);

Error in Titta/calibrate (line 844)
            prevEyeOpennessState = obj.buffer.setIncludeEyeOpennessInGaze(true);
`

As it says, the problem is at line 844, when the calibration routine tries to set the Eye Openness stream; I didn't find any option to disable it.

The way I solved it, I just changed that line with

`
try
                prevEyeOpennessState = obj.buffer.setIncludeEyeOpennessInGaze(true);
            catch ME
                if strcmp(ME.message, 'Titta::cpp::setIncludeEyeOpennessInGaze: Cannot request to record the eyeOpenness stream, this eye tracker does not provide it')
                    prevEyeOpennessState = obj.buffer.setIncludeEyeOpennessInGaze(false);
                else
                    rethrow ME
                end
            end
`

so that if the error is coming from the eye tracker not having the feature, it just sets it to false, otherwise it rethrows the exception.

Thinking about it, this probably belongs better in TittaMex than in Titta though...

Thanks!

dcnieho, 2022-07-28:
Thanks for the catch. Instead, i need to guard the calls with a check
whether the eye openness stream is available (obj.buffer.hasStream('
eyeOpenness')). Stupid this crept through, will fix in the next few days.

ksgfan, 2022-08-08:
I had the same issue with Tobii Fusion Pro. eann's solution works great, thanks!

dcnieho, 2022-08-09:
Hi Both,

Thanks again for the reports! I have not managed to get access to an eye
tracker to test yet, so i have pushed an untested fix to Titta. Its a
little different from eann's solution, i simply check if the eyeOpenness
stream is available before requesting it. As said, i could not test. Could
you test if latest master works for you with the Fusion?

dcnieho, 2022-08-15:
@eann or @ksgfan: can you confirm that the current version works with your fusion without needing to add your workaround?

ksgfan, 2022-08-15:
Yes, the new version works well. Thanks! :)

dcnieho, 2022-08-15:
great, thanks!

closed 2022-08-15 by dcnieho