dcnieho/Titta #40: Linux error on Ubuntu 20.04: glibcxx_3.4.29 not found

iandol, 2022-08-15:
Hi Dee, we are currently still using Ubuntu 20.04 and with the latest Titta (after solving #38), we have the following error:

[image]

Do you build with 22.04? We could recompile Titta on our system as one solution.

Another possible solution is here: https://askubuntu.com/questions/1393285/how-to-install-glibcxx-3-4-29-on-ubuntu-20-04 which suggests to use this PPA: https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/glibc

We will test this fix and update here, but I just wanted to mention this.

dcnieho, 2022-08-15:
Hi Ian,

Yes, I built the latest one on 22.04, so that might be why. I did this
since, if i read and recall correctly, PTB has also moved to recommending
22.04. I'll add a note to the readme. Recompiling yourself should be easy
(just run /TittaMex/makeTittaMex.m), assuming the gcc that you have on your
system is new enough (see
https://github.com/dcnieho/Titta/tree/master/TittaMex).

dcnieho, 2023-01-03:
Hi Ian, Did you resolve this?

iandol, 2023-01-04:
We upgraded to 22.04 on all our systems so yes this is fixed, I'll close...

closed 2023-01-04 by iandol