dcnieho/Titta #41: Understanding the output file

ksgfan, 2022-08-15:
Hi, 

is there any Wiki about the meaning of variables in the output `.mat` file? 

I am having trouble to understand, why there are 3 variables in `dat.data.gaze.left.gazePoint.inUserCoords` and what is their meaning (why the correlation between 2nd and 3rd is equal to 1)? What is the difference between `dat.data.gaze.left.gazeOrigin`and `dat.data.gaze.left.gazePoint`? How would you compute fixations and saccades from the data?

Best,
Dawid

dcnieho, 2022-08-15:
Hi Dawid,

What are the 2nd and 3rd variables you are referring to?

The various data points are defined in the tobii documentation here:
https://developer.tobiipro.com/commonconcepts.html. I'll add a note to
the documentation.

As for how to do fixation classification with these output files, see
/demos/readme_analysis_example in the titta distribution.

Cheers,
Dee

ksgfan, 2022-08-15:
many thanks! :)

dcnieho, 2022-08-15:
Hmm, i assume that they are correlated but not the same, correct? That makes sense when the screen plane is slanted with respect to the coordinate system of the eye tracker, then any change in vertical gaze angle (so Y coordinate of gazePoint on the screen in the user coordinate system) will also yield a change in Z coordinate, viewing distance)

ksgfan, 2022-08-15:
Their correlation is 1. 
`corr(dat.data.gaze.left.gazePoint.inUserCoords(2, :)', dat.data.gaze.left.gazePoint.inUserCoords(3, :)', 'rows', 'complete')`
```
ans =

    1.0000
```

dcnieho, 2022-08-15:
Are the values the same? If the values makes sense that's not weird

ksgfan, 2022-08-16:
No, the values are not the same and make sense. Thanks!

closed 2022-08-16 by dcnieho