dcnieho/Titta #45: Testing on macOS?

iandol, 2023-01-09:
Hi Dee, I see you have done some work on compiling for macOS (is that via Github Actions?) Just to say I'd be happy to test, even if it is only the dummyMode, if you don't have any test hardware. While I run all my experiments via Linux, my personal laptop is a macbook...

dcnieho, 2023-01-09:
Super! The main point of all that work is to make a Python wrapper implementing the same interface as my mex file to use under the hood in the python/psychopy version of Titta. Since that would have worked on Macs before, i want it to work there again. I have an M2 mac mini now that i can work with. What processor do you have in yours?A side effect of this is that the mex file now also compiles on OSX. I saw some problems still however: https://github.com/dcnieho/Titta/commit/2d6997d449687df5db588b72e6a408ae6647958f
And i haven't tested the Python version yet, as the Linux wheel packaging is giving me a lot of grief :p

iandol, 2023-01-09:
Ah, my macbook is also M2, though my older Intel iMac is still seeing some use in my office, and has PTB / MATLAB / Titta installed too, so that could be helpful...

dcnieho, 2023-01-15:
Hmm, trying to get full parallel X86 installs going on the arm macs turns out to be hell (for me at least). So i ordered an Intel mac mini as well, now that i could still get one. That'll do the trick. That said, for as far as i couldn't really test, the python lib (pip install TittaPy!) at least loads on OSX and can tell me its version number and connected eye trackers. So thats something.

I wouldn't want to bother you with testing, as my current problems need rapid iteration. Thanks for the offer again tho! OSX compatibility on matlab isn't high on my list, but on Python it is. Our python version of Titta is moving over to this backend, and since the old version had mac support (being pure Python code), the new one should too...

dcnieho, 2023-01-18:
Ok, on the Intel mac the eye images work like normal, but i still have the issue with the timestamps being systematically off, and crash upon clear all after i have ran my readme (just init()) is not enough. I won't look further into this for now. But lets keep this open as a record of the Mac OSX trouble. This is on 13.1 by the way.