dcnieho/Titta #55: Question concerning calibration data

attentionaffectlab, 2023-05-22:
Hi all,

I'm Eoghan McCauley, the lab manager at the Attention and Affect Lab at the University of Tübingen. We've just got a couple of questions with respect to how calibration works in Titta, and are specifically interested in finding out if/where the calibration data is saved and if we can access it. For reference our device is the Tobii Pro Nano, and we have set it up using the Tobii Pro Eye Tracker Manager. 

Our first question is with respect to the function calibration plays in recording eye data - we would like to know if running the calibration sets parameters which are then used in real time when recording eye data, or if the calibration defines parameters which are or can be used to transform the eye data *after* the recording has been made. We understand that this is really a question for Tobii, and have contacted them as well, but just wanted to confirm the exact procedure used by the Titta toolbox.

Our follow up question is then with respect to whether or not the calibration data is saved, and if so, where it is saved by Titta and how we should go about accessing it.

If you have any other general advice/information with respect to how the calibration works and the calibration data then we would be very interested to know.

Please don't hesitate to contact me at attentionaffectlab@gmail.com if you have any questions.

Hope you are well!

Eoghan

dcnieho, 2023-05-22:
Hi Eoghan,

Thanks for interest in using Titta!

Where do the questions come from/what problem are you solving?

The current Tobiis work with an eye model that help it transform features such as pupil and corneal reflections in the eye image to gaze direction. The calibration figures out the best parameters of this model for a participant, and is then immediately applied for all further recorded data. Tobii's model is proprietary and thus can't be applied afterwards. Nor is the data it likely uses as input (since it proprietary i don't know the details) available. Only calibrated data is available. If you do not calibrate at all, you get data using their eye model with some default parameters. Thats likely not a good fit to an individual subject.

The calibration data is stored by Titta in the mat file, but only in some proprietary format from Tobii. You (and I) cannot do anything with it, other than loading it in again if you want to reuse a calibration of a participant for a later run of your experiment instead of recalibrating. See the documentation of the `previousCalibs` parameter of `Titta.calibrate` for that.

The most important advice i can give for data quality with the Nano is: 1) always calibrate and 2) use a chinrest. It does not deal well with head movement, especially towards and away from the screen. To see that, calibrate with Titta and then on the calibration result screen turn on show gaze. Keep looking at one location on the screen while you move your head and see where the eye tracker thinks you're looking--it should be way off at times.

Cheers,
Dee

attentionaffectlab, 2023-05-22:
Hi Dee,

Thanks so much for your very informative and useful reply - I just have one
follow up. When we run the readme.m script in demos to run a calibration,
do those calibration values remain the same for all subsequent recordings
using the eyetracker until we recalibrate?

Thanks so much again for your help!
Eoghan

dcnieho, 2023-05-22:
Hi Eoghan,

Yes, a calibration remains applied as long as you don't enter calibration
mode again. You can even go to the first screen with the head position of
Titta and then skip (shift+s) and your old calibration will remain active.
Only when you start a new calibration (space on that head position display)
does the previous one get cleared. Or if you switch off the eye tracker
somehow of course.

Cheers,
Dee

> Message ID: ***@***.***>
>

attentionaffectlab, 2023-05-22:
Sorry to send so many emails but unfortunately I have another clarification
I'd like to make -

This sounds silly but what exactly do you mean when you say switch off the
eyetracker? Does this just mean when you stop recording data or something
more specific (like running Etrhndl.delnit();)?. We're interested in
knowing whether or not if, when you run a calibration, those values
become in some way the default calibration values, such that e.g. were I to
run a calibration and then make a recording, then shut down the PC, restart
it and then immediately try and make another recording (without running
another calibration), the recording would still be using the calibration
values from that first calibration? Or would the values at that point be
reset i.e. after restarting the PC would the calibration values return to
their Tobii defaults, such that they can only be altered by loading in the
old calibration or by making a new one?

What we are specifically interested in doing is running a calibration for a
participant using a standalone script and then separately running our task
(whilst recording pupil data with the eyetracker) with another script, as
we are having some difficulties integrating everything together, and it
would be substantially easier for us if we could somehow set the
calibration values for a participant in advance before running the task.
From your last email it looks like whenever the eyetracker has been
switched off (as it would be between our running the two scripts) the
calibration values are also reset, such that we wouldn't be able to simply
run the calibration script, have it finish and then run our task script and
record data using those same calibration values. But it *should* be
possible for us to run a calibration script, save those values and then
load those values for our task script and then use them for that recording.
Is that correct? I'm sorry if any of this is confusing, but I just want to
make sure I really understand what is going on.

And if this plan to save the calibration values and then load them
separately is possible, do you have any advice for how we could go about
loading those values from the mat file before a recording?

Again, thanks so much for all of your help with this, we really do
appreciate it. I understand that my questions here are very specific and
potentially confusing so don't be afraid to clarify things.

Hope you are well!
Eoghan

dcnieho, 2023-05-22:
Ah, switch off is vague indeed, especially for a Nano which doesn't have an
off switch. I'm not 100% sure how it works with the Nano, but i would
assume that calibrating in one script, stopping that program, and then
running another script should work. You can of course easily test, as the
difference between uncalibrated and calibrated data is likely massive. I
would assume that if the Nano somehow loses power (like unplug, or restart
computer), that then it certainly loses its calibration.

Regarding storing and loading calibrations, see the help of Titta.calibrate
like i pointed you to, and just try it out.

Cheers,
Dee

> Message ID: ***@***.***>
>

iandol, 2023-05-24:
Loading and saving calibrations is super easy and flexible with Titta, I save each calibration with a subject name and date by default, then can load that calibration again for a new session. It is as easy as just saving and loading a variable, for an example from some production code loading the mat and then if it is a valid calibration applying it with `.calibrate` or `.calibrateManual` Titta functions: https://github.com/iandol/opticka/blob/master/eyetracker/tobiiManager.m#L282

closed 2023-07-17 by dcnieho