dcnieho/Titta #76: Export to CSV?

qwasium, 2024-02-23:
Hello, I'm a master's student working under Prof. Ippeita Dan.
We have been using Titta with Tobii Pro Spectrum on Ubuntu.

We use Python and R for analysis, and we needed to export the outputed struct to csv or txt or something.
I have created an overcomplicated script for this, (https://github.com/qwasium/export-Titta-mat) and I will be happy to open a pull request if you are interested in merging it, but I feel this sort of thing should be done directly to the C source code.
I know it's not the MATLAB way, but it would be nice to hava a more clean way to output to csv or txt or something universal.

I will be graduating next month so I will be losing my MATLAB license and our lab is moving to Windows and PsychoPy.
So, unfortunately, I don't have time to work on this, but just wanted to shout out some feedback before we move on.

Finally, Titta has contributed to our research greatly and I would like to show my gratitude.
Thank you for creating and maintaining the project.
We are planning to get a paper out citing Titta sometime soon.

Simon

dcnieho, 2024-02-23:
Thank you for your work and your interest in Titta! I briefly looked through your code, and it seems like a good approach (though it indeed got a bit long--not sure it can be avoided). In principle it would be very good to have an option to output to csv. Python Titta also has a way to export its hdf5 files to csv. For now you can read the mat files with scipy (and maybe some other tools), but thats not quite ideal, and only a solution for Python of course.

I would not want to do this at the C++ side, doing it on the matlab side gives more flexibility to users who may need something close but different, and is easier to maintain. I'll look into it. My first aim would probably be to be compatible with Python Titta's export, to keep it easier for end users so they can switch between the two Tittas with less cost.

Thanks!

qwasium, 2024-02-25:
Matching the format to PsychoPy's output would have been an obvious choice. I just started looking into TittaPy so I never had that idea at that time. That was quite dumb of me looking back in hindsight. Thank you for taking the time checking my repo though.
Anyways, I think many would appreciate an official implementation of a csv output. At least, past me would have greatly appreciated it. Thank you for your commitment.
I don't know how many will see this issue, but if anyone feels this is worth the development work, please vote a thumbs up to the above post.

dcnieho, 2024-02-29:
Ok, matching the output for PsychoPy Titta wasn't useful, as that is an ad hoc output for the included analysis examples. I misremembered.

That said, I have now added Titta.saveGazeDataToTSV which you can use to save your gaze data to TSV. Since its a static function, you can just load a Titta mat file and pass the resulting struct to this function. It only saves Gaze and messages to tsv, other data is too complicated as not all fits nicely in a tsv.

To store everything, there is now also saveDataToParquet, which uses Apache Parquet files. These can be read with Pandas, and assume other tools also, and (together with json for some of the info) now contain all the data in the mat file. Also static, so you can also use it to convert all your mat files.

closed 2024-02-29 by dcnieho