dcnieho/Titta #78: settings.UI.val.bgColor leaks to subject display?

iandol, 2024-03-01:
I notice that if I use a dark color like 0.1 for `settings.cal.bgColor` then **after** calibration and validation the screen flashes back to `0.5`, this appears due to `settings.UI.val.bgColor` which is set for subject and operator screens (when the subject see "Please wait...") -- this causes a luminance shift and pupil size change, so if we need to revalidate then the pupil size changing back may affect the revalidation / recalibration for first point a bit. For the subject screen, I think there should only be 1 background color...

iandol, 2024-03-01:
Line 3739 set to:

```matlab
bgClr               = obj.getColorForWindow(obj.settings.cal.bgColor,wpnt(1));
```

fixes this flash. Note this also happens with plot, and line 7082 fixes that:

```matlab
bgClrP          = obj.getColorForWindow(obj.settings.cal.bgColor,wpnt(1));
```

There is also settings.advcal.bgColor, and I suspect if this is different to cal.bgColor then these fixes will not work there. This could be simplified if the options for the subject screen was set in one place: `settings.subject.bgColor` or something?

dcnieho, 2024-03-01:
@iandol: Thanks for the report! I was not fully able to replicate (didn't see a flash with my example code), but i have now hopefully caught the places where with an operator screen the wrong background was used (it should always be `obj.settings.cal.bgColor`). I did not look at `advcal`, yet. It also brought up a point: when having two screens, you should be able to configure the colors for the operator interface and whats on the subject screen separately. So also for instance the setup instruction text and the head color. But thats not possible now.

I'll get back to this, let me know if you can use it for now.

iandol, 2024-03-02:
I'll test when I get back to my Spectrum on Monday, thanks dee!

dcnieho, 2024-03-03:
@iandol: I've taken a different approach. You now have settings.UI.operator with some settings for the operator screen during the setup, calibration/validation and validation result screen phases. That is now decoupled from the participant screen. Luminance of the participant screen is set using settings.UI.setup.bgColor, settings.cal.bgColor and settings.UI.val.bgColor during the three phases. There should not be any flash if you set all three of these to what you want. The bug with wrong participant screen luminance when opening the plots should still be fixed (luminance will now be settings.UI.val.bgColor, like for the rest of the validation result screen).

There is nothing to do for the advanced calibration screens, there the luminance for the two screens was already controlled separately in the settings. At least, there were bugs there with the wrong one being used, but at least no new settings have been added :p

closed 2024-03-03 by dcnieho