dcnieho/Titta #100: Small enhancement, masked video player

iandol, 2024-10-24:
Hi, Dee, I made a small tweak to your VideoPlayer:

[image]

I used a Procedural smoothed disc (with an inverted mask so the center is transparent and edge opaque) to mask the video to a circle. I think a circle is generally better for cal/val reliability.

```matlab
obj.masktex = CreateProceduralSmoothedDisc(wpnt,500, 500, [], 250, 60, true, 2);
```

My modified code is here:

https://github.com/iandol/opticka/blob/dev/eyetracker/tittaAdvMovieStimulus.m

Basically I add a property `obj.doMask` and if true then if `obj.masktex` is empty create it and draw it after the videoframe with the same rect and background color. For `setCleanState` also need to close it if it is open. If you are happy with this I'll make a pull request.

dcnieho, 2024-10-25:
Thanks for the contribution! I've merged it and made some changed to it (most importantly, didn't hardcode the mask color but using the background color).
I am wondering: should the circle be enabled by default, since its not what we describe in the paper? I'm leaning to yes, haven't done it for now though. What do you think?

iandol, 2024-10-26:
i think default to yes is better. Using zenodo you can link a DOI to a specific Github release to ensure the code in the paper is uniquely identified, but otherwise if there is no version information then I don't think we should expect a package to not be developed beyond a publication.

dcnieho, 2024-11-06:
default is set to on now, thanks again!

closed 2024-11-06 by dcnieho