dcnieho/Titta #97: Crash Advanced Calibration: Invalid moviehandle

iandol, 2024-10-24:
Using `readmeAdvancedCalibration_nonhuman_primate.m`

Every so often I get:

```
Error in function OpenMovie: 	Usage error
Invalid moviehandle provided. No movie associated with this handle !!!

...

Error using Screen
Usage:

[ moviePtr [duration] [fps] [width] [height] [count] [aspectRatio] [hdrStaticMetaData]]=Screen('OpenMovie', windowPtr, moviefile [, async=0] [, preloadSecs=1] [,
specialFlags1=0][, pixelFormat=4][, maxNumberThreads=-1][, movieOptions]);

Error in VideoPlayer/openNextVid (line 197)
                [vpnt,vdur] = Screen('OpenMovie', obj.wpnt, char(obj.videos(id)), 0, 1, 2);
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error in VideoPlayer/start (line 75)
                [obj.playingVid, obj.playingVidDur, obj.vidIndex] = obj.openNextVid(false);
                                                                    ^^^^^^^^^^^^^^^^^^^^^^
Error in readmeAdvancedCalibration_nonhuman_primate (line 187)
    vp.start();
```

Some cleanup state causes this intermittant error. I don't quite understand it as `OpenMovie` doesn't accept moviehandles but filenames? If you run `readmeAdvancedCalibration_nonhuman_primate.m` immediately after then no crash.

dcnieho, 2024-10-24:
Hi Ian, strange. Could you add a `disp(char(obj.videos(id)))` right in front of line 197 so we can see what comes by there?
Do i understand correctly that this happens on the first run of `readmeAdvancedCalibration_nonhuman_primate.m` after it crashed for some other reason?

iandol, 2024-10-24:
Right bug #98 helps to trigger this one reliably. Spitting out both wpnt and obj.videos(id)

```
disp('DEBUG: ')
disp(obj.wpnt)
disp(char(obj.videos(id)))
```

On first run without crash:

```
DEBUG: 
    10

/home/cog/Videos/0001-0250R.mp4
```

On second run after the tick crash:

```
DEBUG: 
    10

/home/cog/Videos/0001-0250E.mp4
Error in function OpenMovie: 	Usage error
Invalid moviehandle provided. No movie associated with this handle !!!
```

The windowpointer 10 is valid and the file exists.

I suspect this is a PTB problem somewhere. Your original script uses `clear all` and if I do that after the tick crash (but before run again) I get a full MATLAB crash:

```
>> clear all
Error in function PlayMovie: 	Usage error
Invalid moviehandle provided. No movie associated with this handle !!!

Abnormal termination:
Segmentation violation

Current Thread: 'PTB mainthread' id 133642514859584
```

A `clear all` shouldn't crash matlab (2024b, Ubuntu 22.04)...

iandol, 2024-10-24:
I tried to repro the crash using `SimpleMovieDemo` then a `clear all` but that doesn't cause any issues.

dcnieho, 2024-10-24:
I can't repro with #98 crash or any crash. I do have another issue with my code that sometimes it just crashes (matlab!) off the bat when trying to load the first video. I need to run `SimpleMovieDemo` first once, and then my code works reliably. I have not been able to figure this out. But i am apparently doing something that gets something in a weird state. I'll try to dig (may/will take a while!) what the exact order of calls is that causes this behavior.