dcnieho/Titta #99: Crash Advanced Controller (custom): crash from auto to calibrate

iandol, 2024-10-24:
This is a bit harder to debug. I am integrating advanced calibration into Opticka. Our reward system is different, in that an arduino handles the timing so all we need is:

```matlab
a=arduinoManager;
open(a);
giveReward(a); % arduino times a TTL pulse ON/OFF asynchronously
```

I've had to modify your `NonHumanPrimateCalController` to this:

https://github.com/iandol/opticka/blob/dev/eyetracker/tittaAdvancedController.m

Your rewardProvider is now this:

https://github.com/iandol/opticka/blob/dev/eyetracker/tittaRewardProvider.m

When run from opticka, if I first:

1) Press a to go into auto mode (no subject so it never advances).
2) Press x to skip training

```
Index in position 1 exceeds array bounds.

Error in tittaAdvancedController/calibrate (line 787)
            calPos = obj.calPoss(obj.calPoint,:).*obj.scrRes(:).';
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error in tittaAdvancedController/tick (line 259)
                            commands = obj.calibrate();
                                       ^^^^^^^^^^^^^^^
Error in Titta/doAdvancedCalib (line 5962)
                        autoCommands = controller.tick();
                                       ^^^^^^^^^^^^^^^^^
Error in Titta/calibrateAdvanced (line 1309)
            out = obj.doAdvancedCalib(wpnt,out,currentSelection,controller);
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error in tobiiManager/trackerSetup (line 398)
				cal = me.tobii.calibrateAdvanced([me.screen.win me.operatorScreen.win], incal, ctrl);
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error in runExperiment/configureEyetracker (line 2055)
				trackerSetup(eT);
    ^^^^^^^^^^^^^^^^
Error in runExperiment/runTask (line 710)
				configureEyetracker(me, s);
    ^^^^^^^^^^^^^^^^^^^^^^^^^^
Error in opticka_ui/OKStartTask_Callback (line 1172)
				runTask(app.o.r);
```

iandol, 2024-10-24:
In fact I missed that I had to explicitly call `setCalPoints` and `setValPoints` before calibration on the advanced controller. I suspect there is no way to use some safe defaults here, if so I'll close this.

dcnieho, 2024-10-25:
I don't think there is a way to set save defaults indeed, so lets close this.

closed 2024-10-25 by dcnieho