dcnieho/Titta #102: No connection to TobiiProLab on second computer

ninaalisahinz, 2024-10-25:
Hello team,

we have a two-computer-setup with Matlab & Titta being on one computer (Linux, Ubuntu 24.04), and TobiiProLab, the Tobii Eye Tracker Manager on the other computer (Windows 10 Educational). The eyetracker is also connected to the computer with TobiiProLab.

When trying to establish a connection with talkToProLab we always get the following error
    >> TalkToProLab('Project28', true, '134.100.144.165')
    Error using assert
    TalkToProLab: Could not connect to clock service, did you start Tobii Pro Lab and open a project?
    
    Error in TalkToProLab (line 64)
    assert(this.clientClock.Status==1,'TalkToProLab: Could not connect to clock service, did you start Tobii Pro Lab and open a project?');

We run a lot of diagnostics and
- TobiProLab opens the ports on the localhost and the network IP
- In the first 30 seconds or so you can see that the network TobiiProLab-port is connected to the other port. The connection breaks after some seconds. (But the error message pops up immediately, so it is not as if the connection is not stable and this is why we get the error message).

So it cannot be due to the port being busy or not open. It can also not be due to the different operating systems as we also get the same message when running the command from another windows computer.
If running Titta from the TobiProLab-computer the connection is established and we can run the example experiment.

So I guess it is due to Titta not being able to connect to the client services but we have no idea why.
Has anyone ever used Titta in a two-computer-setup and maybe done some magic to make it work?
Or does anyone have an idea of further diagnostics/workarounds we could try?

Very thankful for any help!
Nina

ninaalisahinz, 2024-10-25:
P.S. also deactiving the clock service in the TalkToProLab command (so TalkToProLab('Project28', false, '134.100.144.165') unfortunately gives the same error.

dcnieho, 2024-10-25:
Hi Nina,

Thank you for your interest in Titta and this functionality! Running a two computer setup has never been tested and is not officially supported. I am surprised you do not run into this error https://github.com/dcnieho/Titta/blob/master/talkToProLab/TalkToProLab.m#L125, and see also the documentation [here](https://github.com/dcnieho/Titta?tab=readme-ov-file#construction-1) and on line 31 of TalkToProLab.m.

What version of Pro Lab are you using? There was a bug in 1.207.44884 where it would refuse remote connections completely, but it seems you get past that at least, so i guess that is fixed.

To test, can you run the following (if Python is ok):
```python
import logging
logger = logging.getLogger('websockets')
logger.setLevel(logging.DEBUG)
logger.addHandler(logging.StreamHandler())
 
import asyncio
import websockets
 
async def hello():
    async with websockets.connect("ws://127.0.0.1:8080/record/externalpresenter?client_id=RemoteClient") as websocket:
        await websocket.send("Hello world!")
        await websocket.recv()
 
asyncio.run(hello())
```
changing the localhost ip to the one of the other machine and see if the connection is properly established (It'll error as this code sends a message that is not understood, but that is ok).

If it is not possible for you to run a single computer setup (I'd recommend that), I'd be happy to guide you to make it work with two computers. But note the trouble from the documentation that we'll have to implement time sync logic needed to convert stimulus computer timestamps to the time of the remote computer running pro lab.

I am just off on holiday for a week. Looking forward to hearing how you plan to continue.

ninaalisahinz, 2024-10-29:
Thanks for the quick reply!

We do have version 1.207.44884...
But we can make a connection to it when using Titta on the same computer. Does this mean it is allowing remote connections, remote here meaning connections from other programs in the external presenter mode? Or do you mean remote = from another computer?
So do you think the issue might solve with updating Pro Lab?

Anyway, I ran the python-script you asked me to run and it does give the following error message

> = connection is CONNECTING
> > GET /record/externalpresenter?client_id=RemoteClient HTTP/1.1
> > Host: 134.100.144.165:8080
> > Upgrade: websocket
> > Connection: Upgrade
> > Sec-WebSocket-Key: GP/QV87UIqVPXM+Ofcbssg==
> > Sec-WebSocket-Version: 13
> > Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
> > User-Agent: Python/3.12 websockets/10.4
> < HTTP/1.1 400 Bad Request
> < Content-Type: text/html; charset=us-ascii
> < Server: Microsoft-HTTPAPI/2.0
> < Date: Tue, 29 Oct 2024 16:08:09 GMT
> < Connection: close
> < Content-Length: 334
> ! failing connection with code 1006
> x half-closing TCP connection
> = connection is CLOSED
> Traceback (most recent call last):
>   File "/home/epsy/Downloads/checkWebsocket.py", line 29, in <module>
>     asyncio.run(hello())
>   File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
>     return runner.run(main)
>            ^^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
>     return self._loop.run_until_complete(task)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
>     return future.result()
>            ^^^^^^^^^^^^^^^
>   File "/home/epsy/Downloads/checkWebsocket.py", line 21, in hello
>     async with websockets.connect("ws://134.100.144.165:8080/record/externalpresenter?client_id=RemoteClient") as websocket:
>   File "/usr/lib/python3/dist-packages/websockets/legacy/client.py", line 642, in __aenter__
>     return await self
>            ^^^^^^^^^^
>   File "/usr/lib/python3/dist-packages/websockets/legacy/client.py", line 659, in __await_impl_timeout__
>     return await asyncio.wait_for(self.__await_impl__(), self.open_timeout)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
>     return await fut
>            ^^^^^^^^^
>   File "/usr/lib/python3/dist-packages/websockets/legacy/client.py", line 666, in __await_impl__
>     await protocol.handshake(
>   File "/usr/lib/python3/dist-packages/websockets/legacy/client.py", line 332, in handshake
>     raise InvalidStatusCode(status_code, response_headers)
> websockets.exceptions.InvalidStatusCode: server rejected WebSocket connection: HTTP 400
Does this make any sense to you?

We would love to do a two-computer setup as actually PTB is running awfully on the single computer that is running Pro Lab right now, and we would prefer to record the data right in Pro Lab to not manually code data recording. So it would be great if you could help us with that!

dcnieho, 2024-11-01:
Hi Nina,

That is indeed a sign that the connection is refused as I've seen before. Please try to update to the latest Pro Lab if you can, and see if that helps. With that info in hand I will reach out to Tobii. However, if the latest version doesn't help, it ma take a while for it to get fixed...

dcnieho, 2024-11-01:
Hi Nina,

I just did a quick test with the latest Pro Lab (24.21.435), and it is now possible to connect from a remote computer. As said, there is still a need to implement time sync logic needed to convert stimulus computer timestamps to the time of the remote computer running pro lab. I also realize that the setup may be a little complicated in terms of cabling as both the computer running Titta and the computer running Pro Lab need access to the eye tracker. So that would only work with eye trackers with a network interface, where you could connect the two computers and the eye tracker together with a switch. Which eye tracker do you have?

All the best,
Dee

ninaalisahinz, 2024-11-01:
Hey Dee,

thanks for the answer and test!
I also did a quick test in our lab with the Tobii Pro Lab-version previous to ours (we have not subscribed to further updates, so I cannot test newer versions) and your python-script seemed to be able to establish a connection at least this time:
= connection is CONNECTING
GET /record/externalpresenter?client_id=RemoteClient HTTP/1.1
Host: 134.100.144.165:8080
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Key: SI7LnY5GwGSX4ctRxxDZ/g==
Sec-WebSocket-Version: 13
Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
User-Agent: Python/3.12 websockets/10.4
HTTP/1.1 101 Switching Protocols
Server: websocket-sharp/1.0
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Accept: 3N0uavcaGH2S25QrccXEmbCyrYY=
Sec-WebSocket-Extensions: permessage-deflate; client_no_context_takeover; server_no_context_takeover
connection is OPEN
TEXT 'Hello world!' [12 bytes]
! failing connection with code 1006
= connection is CLOSED
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/websockets/legacy/protocol.py", line 968, in transfer_data
    message = await self.read_message()
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/websockets/legacy/protocol.py", line 1038, in read_message
    frame = await self.read_data_frame(max_size=self.max_size)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/websockets/legacy/protocol.py", line 1113, in read_data_frame
    frame = await self.read_frame(max_size)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/websockets/legacy/protocol.py", line 1170, in read_frame
    frame = await Frame.read(
            ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/websockets/legacy/framing.py", line 69, in read
    data = await reader(2)
           ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/streams.py", line 750, in readexactly
    raise exceptions.IncompleteReadError(incomplete, n)
asyncio.exceptions.IncompleteReadError: 0 bytes read on a total of 2 expected bytes

However, Titta is throwing the same error still:
Error using assert
TalkToProLab: Could not connect to clock service, did you start Tobii Pro
Lab and open a project?

Error in TalkToProLab (line 64)
            assert(this.clientClock.Status==1,'TalkToProLab: Could not connect to clock service, did you start Tobii Pro Lab and open a project?');
 
Error using SimpleWSClient/onError (line 57)
Verbindungsaufbau abgelehnt (Connection refused)
(ws://134.100.144.165:8081/clock?client_id=tittamatlab)

Error in WebSocketClient/errorCallback (line 162)
            obj.onError(char(e.message));

Error in WebSocketClient>@(varargin)obj.errorCallback(varargin{:}) (line 101)
            set(obj.ClientObj,'ErrorCallback',@obj.errorCallback);

We have a Pro Spectrum, so I think the cabling would be possible. We can try using the switch for connecting the Eyetracker to both computers next and see if Titta runs then?

ninaalisahinz, 2024-11-01:
Hello Dee,

I just tried with an older Pro Lab version (we have not subscribed to the updates past our version) and Titta is able to establish the connection now!

Thank you for the hint with the cabling though. We use a Pro Spectrum, so I think establishing the connection to two computers will be working. Will try that next week.

Also now Titta did not throw an error for the clock service, however, you are probably right in that we need to sychronize both clocks. Do you have any ideas on how?
I was reading a little bit into all issues here on GitHub, and I found that PTB on Linux (on which Titta is currently running) offers using the monotonic clock, which is also the clock Tobii Pro Lab is apparently using, if I remember correctly. Do you think this could solve the issue?

dcnieho, 2024-11-01:
Cool, progress! Glad they have fixed it. What version did you upgrade to?

The thing is that the timestamps sent to Pro Lab need to be in the time of the Pro Lab computer. So we need to figure out the offset between the clock used for the events (e.g. flips) on the Linux computer and the Pro Lab computer's clock. Probably using the time service.

I also strongly recommend to also locally on the Linux save all the data together with good messages, as shown in the TalkToProLab demo. Better have everything twice and end up not using one than run into trouble later

ninaalisahinz, 2024-11-12:
We downgraded to Version 1.194 and this seems to be working.

Regarding the timing: Right now we are sending triggers with the sendCustomEvent() function but giving as input the system time stamp from the Tobii Pro SDK for Matlab as input. This seems as if the timestamp would align with the timestamps that the eyetracker records as system time stamps.
Does this seem reasonable to you?

ninaalisahinz, 2024-11-12:
Yes, the recorded data includes as many as samples between the triggers as they should, considering the experimental time that has passed between the triggers and the sampling rate.
We were thinking that maybe the system time stamp in the Tobii SDK refers to the timestamp on the eyetracker. The SDK reference is not really clear about that...

However, we do sometimes get the error message that the timestamp is out of bounds, so it's not foolproof yet. Trying to figure out in what situations this error throws.

dcnieho, 2024-11-12:
But do the recorded data line up with the stimuli? run the talk to pro lab demo, making your edits to make it work on two computers and see, if your design doesn't allow you to.

This likely is not the way to go. system time stamp is the timestamp on the local computer, not on the eye tracker. That is device time stamp.
Maybe what you do would work if the clocks of the two computers are exactly synced, but thats risky

closed 2024-11-18 by dcnieho

ninaalisakollakowski, 2025-01-13:
Hey Diederick,

this is also connected to issue #106 - not quite sure where to put it correctly, so just linking the two here.
We upgraded our Tobii Pro Lab to version 24.21.435 as you successfully managed to establish a connection between Titta and this one. However, with our two computer setup this does not work. We again receive the same error message that the clock service could not be connected. Interestingly, we get the same error with every version after 1.194. So something in Pro Lab might have changed after this version that makes connection from another computer impossible?
Or do you think this might be related to your recent changes in throwing the clock error that you discussed with Paul in issue #106 - maybe the two clocks can synchronize only in version 1.194 now but not in more recent versions?
I don't know if you know the answer, but I guess you have an educated guess on this?
I am also contacting Tobii again whether they now anything.

ninaalisakollakowski, 2025-01-14:
@dcnieho When we are calling TalkToProLab with the project name and IP under version 24.21.435 we get the following error:
```
Error using assert
TalkToProLab: Could not connect to clock service, did you start Tobii Pro Lab and open a project?

Error in TalkToProLab (line 62)
            assert(this.clientClock.Status==1,'TalkToProLab: Could not connect to clock service, did you start Tobii Pro Lab and open a project?');

Error in al_infantConditions (line 112)
     talkToProLab = TalkToProLab('Disttrack1 Pilot', '134.100.144.165'); % Project name ersetzen!
```

When doing the same thing with version 1.194 the connection is established and we get the following output in the command window (if this is in any way useful)
```
TalkToProLab: using clock API version 1.0
TalkToProLab: using project API version 1.0
TalkToProLab: using external presenter API version 1.0
TalkToProLab: Connected to Tobii Pro Lab, currently opened project is 'Disttrack3 Pilot' (db8fee04-4328-4cd5-bab6-7734bbce9a57)
ready
```

dcnieho, 2025-01-14:
@ninaalisakollakowski Ah, that has nothing to do with my code changes, its well before that part is reached. Very strange you have this issue. Curious to hear what support says! Also strange you have this issue but it works for me with that Pro Lab version....

ninaalisakollakowski, 2025-01-27:
@dcnieho Unfortunately, the support has no idea (or does not want to open up about it) what might be different between these Pro Lab versions and why it therefore works with one but not the other version. Therefore, I have no idea where to start looking for the solution.
Could you give me some info about the setup you are using, i.e. your OS on each computer, so that I can check if there might be an issue there on our side? Thanks again!

dcnieho, 2025-01-27:
@ninaalisakollakowski As far as i know, there is no relevant difference between these versions, so it is a really puzzling issue. I am running on Windows Server 2022, which is basically Windows 10 for our purposes. The firewall is completely disabled, as is the virus scanner. The computers are directly connected via a network (on the same subnet). I have also tried previously with Ubuntu on one of the machines, also worked fine (I think that was with the latest Pro Lab version, not 100% sure). Could you run that websocket python script from October (see above) both against the working version and the latest failing version and post the output?

dcnieho, 2025-01-31:
It remains very strange. Indeed there is an error building up the connection already. I am not an expert either...
So this is on the exact same machine, everything else the same, just installing different versions of Pro Lab?

ninaalisakollakowski, 2025-02-04:
Yes, everything is the same. In between these two outputs, I was just deinstalling the old Pro Lab version and running the installer of the new Pro Lab version. 
Hm. I will see with our local IT whether they have any more ideas on the issue.
Thanks anyway!

dcnieho, 2025-02-04:
@ninaalisakollakowski: since you have a working solution and i can't for the life of me figure out the problem, i guess we'll leave it here. If the latest pro lab has analysis functionality you need that the older one doesn't have, you can always try to record, then upgrade pro lab, then open the old project and see if you can use all the latest and shiniest analysis functionality ;) On the recording side, nothing has changed anyway as far as i know.