dcnieho/Titta #104: Changing dot size and color in advanced calibration

paul-gallenkemper, 2024-11-08:
Hey, 
(just as a heads up, i'm a colleague of ninaalisahinz)
I wanted to adapt the advanced calibration example code to our infant study. Because i'm afraid the stimuli aren't salient enough, even when moving, i wanted to increase their size and make them colorful. First, i changed the default properties in AnimatedCalibrationDisplay to this: 

        doShrink            = true;
        shrinkTime          = 0.5;
        doMove              = true;
        moveTime            = 1;
        moveWithAcceleration= true;
        doOscillate         = true;
        oscillatePeriod     = 1.5;
        blinkInterval       = 0.3;
        blinkCount          = 2;
        fixBackSizeBlink    = 80;  % Increased from 35 to 80
        fixBackSizeMax      = 100; % Increased from 50 to 100
        fixBackSizeMaxOsc   = 80;  % Increased from 35 to 80
        fixBackSizeMin      = 30;  % Increased from 15 to 30
        fixFrontSize        = 20;  % Increased from 5 to 20
        
        fixBackColor        = [255, 0, 0];   % Red background for fixation dot
        fixFrontColor       = [0, 0, 255];   % Blue front cross on fixation dot
        bgColor             = [0, 255, 0];   % Green background screen color
        
Then, i adapted the readmeAdvancedCalibration.m function to handle a two screen Linux setup (global coordinates). When simply calling this adapted readmeAdvancedCalibration.m, this works fine. The dots are bigger and colorful, the calibrateAdvanced() seems to use these values. 
        
But in my own adapted version, which is VERY close to the code of readmeAdvancedCalibration.m, these adaptations are gone and the usual grey and small stimuli are shown. Additionally, i tried setting the colors in my function. This also does not work. 

I'd say the only differnce between the codes is that i obviously deleted the section showing rabbits and that i passed my own EThndl and screen size instead of setting it up in the calibration. I stepped through the entire code and i cannot identify when my code deviates from readmeAdvancedCalibration. It calls AnimatedCalibrationDisplay() just the same and sets up the CalViz the same. Both my function and readmeAdvancedCalibration call the same instance of tobii.calVal{1} = EThndl.calibrateAdvanced([wpntP wpntO]); seemingly with the same input.

Note that this also happens when i put my calibation function file in the same directory as readmeAdvancedCalibration. 
Note that i have tried not passing the EThndl to my function but setting it up after changing the colors for the calibration, also did not work.

As i cannot add .m files here, this is the version of my code where in which i try to adapt the colors 1. by defining them there and 2. by calling the adapted AnimatedCalibrationDisplay(); and neither changes the color:

function al_infantETCalibration(taskParam, EThndl)

DEBUGlevel              = 2;

fixClrs = [
    255, 0, 0;   % Red
    0, 255, 0;   % Green
    0, 0, 255;   % Blue
    255, 255, 0; % Yellow
    255, 0, 255; % Magenta
    0, 255, 255  % Cyan
];

bgClr                   = 127;
eyeColors               = {[255 127 0],[0 95 191]}; % for live data view on operator screen
useAnimatedCalibration  = true;
scrParticipant          = 0;
scrCoordinatesParticipant = taskParam.display.screensize;
scrOperator             = 0;
scrCoordinatesOperator = [0 0 1920 1080];
useWindowedOperatorScreen = false;  % if true, a windowed operator display smaller than the whole screen is made
% task parameters
fixTime                 = .5;
imageTime               = 4;
% live view parameters
dataWindowDur           = .5;   % s

home = cd;
cd ..;
cd(home);

try
    eyeColors = cellfun(@color2RGBA,eyeColors,'uni',false);
    
    % request some debug output to command window, can skip for normal use
    settings.debugMode      = true;
    % customize colors of setup and calibration interface (yes, colors of
    % everything can be set, so there is a lot here).
    % 1. operator screen
    settings.UI.advcal.bgColor              = bgClr;
    fixBackColorIndex                       = randi(size(fixClrs, 1));
    settings.UI.advcal.fixBackColor         = fixClrs(fixBackColorIndex, :);  
    fixFrontColorIndex                      = randi(size(fixClrs, 1));
    settings.UI.advcal.fixFrontColor        = fixClrs(fixFrontColorIndex, :);  
    settings.UI.advcal.fixPoint.text.color  = fixClrs(1);
    settings.UI.advcal.avg.text.color       = fixClrs(1);
    settings.UI.advcal.instruct.color       = fixClrs(1);
    settings.UI.advcal.gazeHistoryDuration  = dataWindowDur;
    settings.UI.advcal.fixPoint.text.size   = 24;
    % 2. calibration display
   if useAnimatedCalibration
        % custom calibration drawer
        calViz                           = AnimatedCalibrationDisplay();
        settings.advcal.drawFunction     = @calViz.doDraw;
        calViz.bgColor                   = bgClr;

        fixBackColorIndex                = randi(size(fixClrs, 1));
        settings.UI.setup.fixBackColor  = fixClrs(fixBackColorIndex, :); 
        settings.UI.advcal.fixBackColor  = fixClrs(fixBackColorIndex, :);
        settings.UI.val.fixBackColor = fixClrs(fixFrontColorIndex, :); 
        settings.UI.cal.fixBackColor = fixClrs(fixFrontColorIndex, :);
        settings.advcal.fixBackColor = fixClrs(fixFrontColorIndex, :);

        ixFrontColorIndex                = randi(size(fixClrs, 1));
        settings.UI.advcal.fixFrontColor = fixClrs(fixFrontColorIndex, :);
        settings.UI.setup.fixFrontColor = fixClrs(fixFrontColorIndex, :); 
        settings.UI.val.fixFrontColor = fixClrs(fixFrontColorIndex, :); 
        settings.UI.cal.fixFrontColor = fixClrs(fixFrontColorIndex, :);
        settings.advcal.fixFrontColor = fixClrs(fixFrontColorIndex, :);

    else
        % set color of built-in fixation points
        settings.advcal.bgColor         = bgClr;
        settings.advcal.fixBackColor    = fixClrs(1);
        settings.advcal.fixFrontColor   = fixClrs(2);
    end
    % callback function for completion of each calibration point
    settings.advcal.cal.pointNotifyFunction = @demoCalCompletionFun;
    settings.advcal.val.pointNotifyFunction = @demoCalCompletionFun;

    nLiveDataPoint  = ceil(dataWindowDur*EThndl.frequency);

    if DEBUGlevel>1
        % make screen partially transparent on OSX and windows vista or
        % higher, so we can debug.
        PsychDebugWindowConfiguration;
    end
    if DEBUGlevel
        % Be pretty verbose about information and hints to optimize your code and system.
        Screen('Preference', 'Verbosity', 4);
    else
        % Only output critical errors and warnings.
        Screen('Preference', 'Verbosity', 2);
    end
    Screen('Preference', 'SyncTestSettings', 0.002);    % the systems are a little noisy, give the test a little more leeway
    [wpntP,winRectP] = PsychImaging('OpenWindow', scrParticipant, bgClr,scrCoordinatesParticipant, [], [], [], 4);
    if useWindowedOperatorScreen
        wrect  = Screen('GlobalRect', scrOperator);
        [w, h] = Screen('WindowSize', scrOperator);
        wrect  = CenterRect([w*.1 h*.1 w*.9 h*.9],wrect);
        [wpntO,winRectO] = PsychImaging('OpenWindow', scrOperator, bgClr, wrect, [], [], [], 4, [], kPsychGUIWindow);
    else
        [wpntO,winRectO] = PsychImaging('OpenWindow', scrOperator, bgClr,scrCoordinatesOperator, [], [], [], 4);
    end

    hz=Screen('NominalFrameRate', wpntP);
    Priority(1);
    Screen('BlendFunction', wpntP, GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA);
    Screen('BlendFunction', wpntO, GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA);%useWindowedOperatorScreen
    Screen('Preference', 'TextAlphaBlending', 1);
    Screen('Preference', 'TextAntiAliasing', 2);
    % This preference setting selects the high quality text renderer on
    % each operating system: It is not really needed, as the high quality
    % renderer is the default on all operating systems, so this is more of
    % a "better safe than sorry" setting.
    Screen('Preference', 'TextRenderer', 1);
    KbName('UnifyKeyNames');    % for correct operation of the setup/calibration interface, calling this is required
    
    % do calibration
    try
        ListenChar(-1);
    catch ME
        % old PTBs don't have mode -1, use 2 instead which also supresses
        % keypresses from leaking through to matlab
        ListenChar(2);
    end

    tobii.calVal{1} = EThndl.calibrateAdvanced([wpntP wpntO]);
    ListenChar(0);

Screen('Close', wpntO);
Screen('Close', wpntP);  

catch me
sca
ListenChar(0);
rethrow(me)

end
"

would you please help me identify where the colors and sizes defined in AnimatedCalibrationDisplay() are overwritten and set back to standard?

Thanks, 
Paul

dcnieho, 2024-11-08:
Thanks for your interest!

Your code doesn't run as is and can't be what you have been testing as you'd get various error messages from matlab (syntax) and Titta. I had to make some changes, and then it all works fine:
```matlab
DEBUGlevel = 2;

fixClrs = [
    255, 0, 0; % Red
    0, 255, 0; % Green
    0, 0, 255; % Blue
    255, 255, 0; % Yellow
    255, 0, 255; % Magenta
    0, 255, 255 % Cyan
    ];

bgClr = 127;
eyeColors = {[255 127 0],[0 95 191]}; % for live data view on operator screen
useAnimatedCalibration = true;
scrParticipant          = 2;
scrOperator             = 1;
useWindowedOperatorScreen = false; % if true, a windowed operator display smaller than the whole screen is made
% task parameters
fixTime = .5;
imageTime = 4;
% live view parameters
dataWindowDur = .5; % s

home = cd;
cd ..;
cd(home);

try
    eyeColors = cellfun(@color2RGBA,eyeColors,'uni',false);

    % request some debug output to command window, can skip for normal use
    settings = Titta.getDefaults('Tobii Pro Spectrum');
    settings.debugMode      = true;
    % customize colors of setup and calibration interface (yes, colors of
    % everything can be set, so there is a lot here).
    % 1. operator screen
    settings.UI.advcal.bgColor              = bgClr;
    fixBackColorIndex                       = randi(size(fixClrs, 1));
    settings.UI.advcal.fixBackColor         = fixClrs(fixBackColorIndex, :);
    fixFrontColorIndex                      = randi(size(fixClrs, 1));
    settings.UI.advcal.fixFrontColor        = fixClrs(fixFrontColorIndex, :);
    settings.UI.advcal.fixPoint.text.color  = fixClrs(1);
    settings.UI.advcal.avg.text.color       = fixClrs(1);
    settings.UI.advcal.instruct.color       = fixClrs(1);
    settings.UI.advcal.gazeHistoryDuration  = dataWindowDur;
    settings.UI.advcal.fixPoint.text.size   = 24;
    % 2. calibration display

    if useAnimatedCalibration
        % custom calibration drawer
        calViz = AnimatedCalibrationDisplay();
        settings.advcal.drawFunction = @calViz.doDraw;
        calViz.bgColor = bgClr;

        settings.UI.setup.fixBackColor  = fixClrs(fixBackColorIndex, :);
        settings.UI.advcal.fixBackColor  = fixClrs(fixBackColorIndex, :);
        settings.UI.val.fixBackColor = fixClrs(fixBackColorIndex, :);
        settings.advcal.fixBackColor = fixClrs(fixBackColorIndex, :);
        calViz.fixBackColor = fixClrs(fixBackColorIndex, :);

        settings.UI.advcal.fixFrontColor = fixClrs(fixFrontColorIndex, :);
        settings.UI.setup.fixFrontColor = fixClrs(fixFrontColorIndex, :);
        settings.UI.val.fixFrontColor = fixClrs(fixFrontColorIndex, :);
        settings.advcal.fixFrontColor = fixClrs(fixFrontColorIndex, :);
        calViz.fixFrontColor = fixClrs(fixFrontColorIndex, :);

[... 54 lines cut ...]
    catch ME
        % old PTBs don't have mode -1, use 2 instead which also supresses
        % keypresses from leaking through to matlab
        ListenChar(2);
    end

    tobii.calVal{1} = EThndl.calibrateAdvanced([wpntP wpntO]);
    ListenChar(0);

    Screen('Close', wpntO);
    Screen('Close', wpntP);

catch me
    sca
    ListenChar(0);
    rethrow(me)
end
```

Importantly:
1. You missed `settings = Titta.getDefaults('Tobii Pro Spectrum');`
2. You missed `EThndl = Titta(settings);` and `EThndl.init();`. You can't make a settings variable with some settings and not pass it to Titta. Then of course none of your settings take effect. If you want to change settings for an already initialized EThndl, use `Titta.getOptions` and `Titta.setOptions`.
3. I used the default `AnimatedCalibrationDisplay`, so had to add: `calViz.fixBackColor = fixClrs(fixBackColorIndex, :);` and the same for front color. If you don't change the logic of AnimatedCalibrationDisplay, you should not change the function, but instead set the relevant options as i show. Also, around there some front colors are set with fixBackColorIndex.

By the way, for infants you may want to look at readmeImageCal

paul-gallenkemper, 2024-11-08:
Hey, 
this is really helpful already and i think i'm getting a better idea of how Titta works. Also, i have decided to move on to Image Calibration now, as you suggested. 

I think i almost got it now, but it still shows the regular black and white non-animated fixation crosses as if the settings were unchanged. 

In my main function, i set up the Titta instance like this: 

settings = Titta.getDefaults('Tobii Pro Spectrum'); % Use the model from the selected tracker

    % init

    EThndl          = Titta(settings);

    EThndl.init('tet-tcp://169.254.11.151');

    % These settings are important for the calibration procedure

    settings = EThndl.getOptions();

    
    imageFolder = '/home/epsy/Schreibtisch/disttrack/Infant-version-Cannon-Task/Titta/cal_displays';
    images                  = cellfun(@(x) fullfile(imageFolder,x),{'colorWheel.gif','flower.gif'},'uni',false);
    imageFrameDurations     = [repmat(.05,1,30) repmat(.15,1,22)]; % s  (30 frames in first file, 22 frames in second file, and they require different playback speeds, hence this)
    imageScaleFac           = 1;

    % request some debug output to command window, can skip for normal use
    settings.debugMode      = true;

    % customize colors of setup and calibration interface (colors of
    % everything can be set, so there is a lot here).

    % 1. setup screen
    fixClrs                         = [0 255];
    bgClr                           = 127;
    settings.UI.setup.bgColor       = bgClr;
    settings.UI.setup.instruct.color= fixClrs(1);
    settings.UI.setup.fixBackColor  = fixClrs(1);
    settings.UI.setup.fixFrontColor = fixClrs(2);

    % 2. calibration display
    % custom calibration drawer
    calViz                           = ImageCalibrationDisplay();
    calViz.setImages([],images,imageFrameDurations,imageScaleFac);
    calViz.bgColor                   = bgClr;

    % calViz.restartAnimForEachPoint   = false;
    settings.cal.drawFunction        = @calViz.doDraw;

    % NB: initialized further below
    % callback function for completion of each calibration point
    settings.cal.pointNotifyFunction = @demoCalCompletionFun;

    % 3. validation result screen
    settings.UI.val.bgColor                 = bgClr;
    settings.UI.val.avg.text.color          = fixClrs(1);
    settings.UI.val.fixBackColor            = fixClrs(1);
    settings.UI.val.fixFrontColor           = fixClrs(2);
    settings.UI.val.onlineGaze.fixBackColor = fixClrs(1);
    settings.UI.val.onlineGaze.fixFrontColor= fixClrs(2);

    EThndl.setOptions(settings);
    
    
I also tried it without the setOptions() function by simply getting the default Settings, adapting the settings accordingly and initializing Titta using EThndl = Titta(settings) afterwards: 

settings = Titta.getDefaults('Tobii Pro Spectrum');

imageFolder = ...
...

[... 75 lines cut ...]
        ListenChar(-1);
    catch ME
        % old PTBs don't have mode -1, use 2 instead which also supresses
        % keypresses from leaking through to matlab
        ListenChar(2);
    end

    tobii.calVal{1} = EThndl.calibrateAdvanced([wpntP wpntO]);
    ListenChar(0);

Screen('Close', wpntO);
Screen('Close', wpntP);  

catch me
sca
ListenChar(0);
rethrow(me)

end

I feel like the   
      tobii.calVal{1} = EThndl.calibrateAdvanced([wpntP wpntO]); 
 call requires some other input than the Titta instance for the settings to come into effect, but i can't make it out... 
 
Sorry if this is tedious to you. I'm super thankful for your help and really hopeless on my own in front of these huge codes.

closed 2024-11-11 by paul-gallenkemper