dcnieho/Titta #106: Trigger sending on two PC Setup

paul-gallenkemper, 2024-11-13:
Hello again, 
In this next step of our Setup, we want to send triggers to Tobii Pro Lab via talkToProLabs "sendCustomEvent" function. To do so, we currently use: e.g.
talkToProLab.sendCustomEvent(TobiiProSDK.get_system_time_stamp(), sprintf('Block number %d End of Block', block));

meaning we chose Tobii Pro SDKs System Timestamp as the trigger timestamp. This mostly works, but this error is thrown, when the two computers were turned on at two different timepoints. It rarely occurs, when the computers were turned on around the same time (seconds): 

"TalkToProLab: Command 'FinalizeRecording' returned an error: 201 (Recording finalization failed):
Custom event timestamp is out of recording time bounds"

What lead us to chose this timepoint are the TalkToProLab documentation and the documentation here: https://connect.tobii.com/s/article/Why-do-I-have-inconsistent-timestamps-in-the-data-export-file?language=en_US 

In TalkToProLabs documentation, it is stated that "you are responsable to present talkToProLab with timepoints in Pro Lab time, not local time". In Tobiis documentation, it is mentioned that the System Timepoint, depicted as the Computer.timepoint in output tables, is used for Tobii Pro Labs internal timing and is also provided by Tobii Pro SDK (i assume via get_system_time_stamp). I have no idea why this error seems to be thrown at all, because this timepoint seems to me to be the correct one. I also don't know why this error seems to the thrown depending on the synchronization status of the two computer's clocks, when neither of the two functions (sendCustomEvent or get_system_time_stamp) use local time. 

I looked a little bit deeper into possibilities of getting the Tobii Pro Lab time directly into Matlab on the operating PC. It seems that you would either have to build another Websocket Server to tap into the win32 timestamp on the Pro Lap PC or you would have to modify the Websocket client provided by Titta. Neither seems like something i would be comfortable doing. 

Do you have an idea why the Tobii Pro SDK System timestamp might be wrong and what would be a suitable alternative to feed into the sendCustomEvent function? 

Best Regards, 
Paul

I'll add my colleague as well, she has been in contact with you over this topic: @ninaalisahinz

dcnieho, 2024-11-13:
As said in #102 (please read through all of that), you need the system time stamp on the Tobii pro lab computer, not on the stimulus computer. They happen to be similar enough when you switch on the computers at around the same time, but that is not a solution. They'll be off, just not as far that you get that error, but nonetheless data and stimuli/custom events will not line up.

Neither of you suggestions is what is needed. All that is needed is to determine the offset in clocks between the two computers, and to apply that offset. That requires some new code to be implemented, using the clientClock object in TalkToProLab. Do you have a timeline for when you need this to work? Then i can indicate if it is realistic that i may have a chance to look at it.

Also, could you tell me exactly how the two computers and the eye tracker are connected to each other? What is your network setup? What IP do you use to connect to the eye tracker on the two computers? Just so i don't have to figure that out when i work on this, saves time.

paul-gallenkemper, 2024-11-13:
I am familiar with Issue #102, Nina and I have been working on that together. 

I am confused why you think the system timestamp collected by get_system_time_stamp using Tobii Pro SDK (which is btw the same as the System.timestamp saved in the gaze data buffer, which can be viewed by peekN('gaze')) is local to the Matlab PC and not the Tobii Pro Lab PC. All of Tobiis documentations (see message above, but also here: https://developer.tobiipro.com/commonconcepts/timestamp-and-timing.html "system time stamp uses the computer on which the application is running as the source") lead me to think that the timestamp gathered this way is actually local to the Tobii Pro Lab PC and should therefore be compatible with Pro Lab. 

Let's still continue with the assumption that this system timestamp is local to the Matlab PC. 

Regarding our schedule: We originally wanted to start data collection long ago, but now we actually wanted to invite some pilot participants for next Monday as everything apart from the trigger sending and some inconsistencies with PTB is now ready. I don't want to decide anything final without my colleagues and my boss, but let's say this should be ready by the end of next week. If this is not convenient for you, no problem at all! You have been of great help already!

This is our network setup: 
The eyetracker and both PCs are connected to a LAN switch. Nothing else is on the LAN switch, but both PCs are also connected to the universities LAN, which is also connected to the internet. 
The Matlab PC (Linux) has the IP address 134.100.144.154 in the local LAN.
The Tobii Pro Lab PC (Windows) has the IP address 134.100.144.165
The eyetracker has the IP address 169.254.11.151, either given as tet-tcp://169.254.11.151 (on Windows Eyetracker Manager) or as tobii-prp://169.254.11.151 (on Linux Eyetracker Manager).

To connect to Tobii Pro Lab, we use: talkToProLab = TalkToProLab('Project29', true, '134.100.144.165'); (sidenote: I've tried this with DoSynchCheck set to false, doesn't really seem to change anything systematically).

To connect to the ET directly, we use: EThndl = Titta(settings);
EThndl.init('tet-tcp://169.254.11.151')
with the settings predefined for advanced image calibration.

Additionally, we simply use TobiiProSDK = EyeTrackingOperations(); to set up a Tobii Pro SDK instance that can get the system timestamp

paul-gallenkemper, 2024-11-13:
I think next friday would be OK aswell. With the first pilots, we usually only aim to find out whether the stimuli and the timing are suitable. 

Yes, the connection to the eyetracker works! 

I don't think the talkToProLab ever threw any errors once we gave it the correct IP address, independent of the DoSynchCheck input. 

Ok, then moving forward we would have to find a way to get the Pro Lab timestamp into the Matlab PC so that we can account for the delay / offset. How would we do that considering that the talkToProLab.clientClock does not have a function that outputs the current Pro Lab timepoint?

dcnieho, 2024-11-13:
I find it very strange that no sync errors are thrown if dosynccheck is true. clientClock, to my understanding, should be providing the Pro Lab timestamp, so what we would need to sync it all together. You know what, it is probably a bug in my check. Looking at it, it will only signal if the lab time is ahead of the matlab computer time, not when it is behind. Easily tested. This i can figure out tmr already at least.

But yeah, implementing the sync is my problem to figure out, and perhaps chat with Tobii about.

dcnieho, 2024-11-14:
Indeed, my dosynccheck not firing was most likely a bug.

I have now fixed that bug, and also made a start at running two computer setups. Sync logic is implemented and tested when making a new TalkToProLab instance, but i am not using it for converting the timestamps provided to sendStimulusEvent and sendCustomEvent. Ran out of time for this week. Can you try out the new code and see if it crashes? As noted, this makes no observable difference for you, you will still have your original problem

paul-gallenkemper, 2024-11-14:
I can try running the new code later today. But for now, we have decided to chose a workaround for trigger sending. 
We would like to record and save data either in Matlab or using Pro Lab and we'd be thankful for some advise. 
If we chose to record data in Matlab, we would not need Pro Lab at all, not even a second PC. To decide on this, it would be good to know whether the buffer running is a strain on working memory that might impact stimulus presentation? 

If we chose to record data in Tobii Pro Lab, we would like to get rid of the buffer recording in Matlab. A suitable workaround for the timestamp synchronization would be using the device timestamp (eyetracker) as this is available to both systems. We could get this timepoint in Matlab during the stimulus presentation and use it to mark stimulus phases after recording in Pro Lab. Is it possible to get the device timestamp without the buffer running? Also, if we chose this option we would still have to open the buffer for our gaze contingent stimulus display. Does the start(‘gaze’) impact working memory or take long itself so that it might cause a delay in stimulus presentation when called during the presentation loop?

dcnieho, 2024-11-14:
Ah, is that the only reason you want to use Pro Lab? I thought/expected it was for visualization or analysis. Eye tracking data isn't big, so there should be no issue at all. I would always recommend to record in matlab, even if you also record in Pro Lab.

This is one of these things that you should first verify there is actually a problem, before solving it. Almost completely sure there is no problem here, unless you're weirdly low on memory.

Since i have this mostly implemented now, will you see it through to the end with me, provided testing when its done?
Look forward to hearing how my code runs on your machine

paul-gallenkemper, 2024-11-15:
Ok, really glad to hear this! We actually consulted some experts from the research unit we are situated in right now as our project is a special case being the only one implementing gaze contingent stimulus display. The other projects naturally separate data recording and stimulus presentation, using various measures but also pupillometry, and advised us to do the same. But wrapping it all up in a one computer setup is a lot easier of course. 

Still i'm happy to keep testing the changes you will make to TalkToProLab in parallel. As of now, nothing has changed with the new version. The talkToProLab doesn't take true / false as a second input anymore, but also doesn't throw an error when run. Matlab still connects to Pro Lab as an external presenter and finalizing our recording after sending custom events with talkToProLab.sendCustomEvent(EThndl.buffer.systemTimestamp, "trigger name") still throws the error "201: timestamp out of bounds". Looking forward to see whether the time synching will work in the future!

dcnieho, 2024-11-18:
Thanks for the test. That is as expected with what i implemented, and that you don't see an error from TalkToProLab means that sync seems to be working fine. I have now implemented the last bit and tested it successfully here with the example provided in Titta. If you run TalkToProLab and provide the address of Pro Lab on another computer, all local timestamps (should be PTB timestamps as per the rest of Titta) will be automatically converted to timestamps of the remote pro lab.

This sync should be very close to perfect (< 1 ms on a well working network), but may not be completely perfect. In testing, one from the five runs i made did fail with out of bounds. So please test a few times. Also as a first test, simply run the readmeprolab scripts, edited to provide the remote pro lab, to ensure error you see are not in your implementation.

Thanks!

paul-gallenkemper, 2024-11-20:
I've tested the sendCustomEvent function using Titta.sendCustomEvent([ ], "name of the trigger"). 
I ran the experiment code a couple of times and it seems just as before: When connecting to Pro Lab on the other PC, no error is thrown. When finalizing the recording, the timestamp out of bounds error is still thrown unfortunately. I was worried about the GetSecs() format being different on the Linux PC: This returns a 10 digit number, whereas the System Timestamps are 12 digit numbers. I saw that the sendCustomEvent() function adds six zeros to the timestamp. Removing this still does not change anything about the error thrown. 

Sorry i couldn't run the readmeProLabIntegration script on our setup now. I don't have the time right now to adapt this code to a two screen calibration, two PC setup. If this information is necessary for you to judge your code's performance, i'll see that i can find some time next week.

dcnieho, 2024-11-20:
Thanks for the test! Hmm, that is a complication i forgot about yeah. I should be using `Titta.getTimeAsSystemTime` instead that deals with this complexity on Linux. Will get back to you later today.

Probably you do not have to adapt anything in that demo script, except set the eye tracker's screen number on line 36 so that Titta shows up on the eye tracker's monitor and the second monitor is unused, and add the remote PC ip address on line 96.

dcnieho, 2024-11-20:
done, please test again

paul-gallenkemper, 2024-11-20:
Regarding the readmeProLabIntegration: unfortunately, Linux doesn't work with screen numbers, but with global coordinates. I have adapted everything as in my (working) adaptation of the calibration procedure and spend around 30min debugging, but i cannot figure it out now. I'll try that next week

Regarding my script: The warning 
"Warning: Matrix is close to singular or badly scaled.
Results may be inaccurate. RCOND =  1.355122e-16. 
> In Synchronizer/doSync (line 59)
In TalkToProLab (line 109)
In..."
Is now thrown when calling talkToProLab, but the finalizeRecording() function works fine. later today i will compare synchronisation via trigger sending or via the device timestamp and check against intended trial phase durations. As of now, i think this looks promising.

dcnieho, 2024-11-20:
Ah, ok, thanks for giving it a try.
In the end, if things work correctly (and also look correctly! hence my request to try my demo code) in pro lab, then we're good.

Hmm, that warning is nothing to worry about, but also not ideal. I'll try an make it a little less likely by changing the sync at start up. Sync is improved during the run anyway, and I assume you didn't see the warning again. 

Looking forward to hearing about your other findings!

paul-gallenkemper, 2024-11-20:
Right, the warning always comes up during the startup of talkToProLab, but never again after that. 

Also, i now managed to get your example code running (our Setup somehow has a problem with "clear all" and keeps crashing on that command...), which now runs error free and as expected.

To conclude, we now gather device and system timestamps and save them in matlab, we send triggers via the synched system timestamp and record data in matlab and in pro lab. I'll get back to you soon with a control of how the timestamps overlap. 

Thank you very much so far :)

dcnieho, 2024-11-20:
Go to hear it seems to be working! Looking forward to the results of that
check.
Does the warning still come up after the change i made?

iandol, 2024-11-21:
@paul-gallenkemper 

> unfortunately, Linux doesn't work with screen numbers, but with global coordinates.

This is not correct (or at least, not optimal). The correct way to set up multiple displays is the following:

https://cogplatform.github.io/Notes/src/PTB/2-SCREENS-PTB.html

Once set up screen 0 is the operator screen and screen 1 is the subject screen (which is different to Windows, I also often have to renumber the values in the Titta demos).

paul-gallenkemper, 2024-11-22:
Yes, the warning still comes up when calling talkToProLab. 

But my results for comparing different timestamping methods look very good. I timestamped the datasets in multiple ways: 
1. For the output table in Tobii, i used the device timestamp recorded at stimulus presentation using peekN() and .deviceTimeStamp
2. For the output table in Tobii, i used trigger sending at stimulus presentation via sendCustomEvent()
3. For the output table in Matlab, i also used the device timestamp recorded at stimulus presentation (redundant to 1.)
4. For the output table in Matlab, i used the system timestamp gathered using buffer.SystemTimeStamp

My observations are: 
- The System Timestamp drawn in Matlab does not overlap with the Computer.timestamp from Tobii. I don't know whether your synching intended this, but I'll just note it here
- The trial phase lengths were mostly within 1ms of the expected length, for all four calculations
- The trial phase startpoints were mostly within 1ms apart from each other, , 5ms at max, across all four timestamping methods.

closed 2024-11-22 by dcnieho