dcnieho/Titta #107: Some Feature Requests / Questions for Infant Advanced Calibration

paul-gallenkemper, 2024-11-20:
Hello again :) 
We have some questions or feature requests regarding the advanced image based calibration that we want to use on infants. 

1. Would it be possible to play sounds during the presentation of the calibration dots?
2. Would it be possible to display the calibration results without validating the calibration? To my understanding, the precision and accuracy of the calibration can only be shown after the validation. 
3. Do you think the validation is neccessary at all? If yes, is it possible to reduce the number of validation points from 4 to 2? 

Best regards, 
Paul + Nina

dcnieho, 2024-11-20:
1. I suggest using video files as calibration stimuli, see https://github.com/dcnieho/Titta/blob/master/demo_experiments/readmeVideoCal.m. Those can have sound (at least that should work, if not, email me a file with sound and i'll have a look).
2. on the advanced screen, as soon as you have collected some calibration data and press the compute calibration button, you'll get an output of the calibration, in form of the lines around the points. Is that what you mean? There are no data quality values (like accuracy) attached to these lines. Tobii doesn't provide enough info for that data to compute it, and even if they did, it would not make sense. Validation is what you need to assess data quality.
3. Validation is fully configurable. The points shown for calibration and validation are set with `settings.advcal.cal.pointPos` and `settings.advcal.val.pointPos`, respectively. These are the points that will be available in the interface. It is then up to the operator to decide which to show and use for calibration/validation. So you can choose two validation points. I'd say yes, validation is always necessary, as you need to report data quality actually measured from the participants (See https://link.springer.com/article/10.3758/s13428-023-02187-1). But in principle, one point can be sufficient for that. For calibration, with a spectrum, I think you'd want at least three points if you can, else two, along a diagonal of the screen, equidistant from the center. You'll need to experiment a bit with that to see what works robustly.

dcnieho, 2024-12-02:
I have now implemented volume control for the video player (well, you set it once when constructing the object). See the change in `readmeVideoCal.m` here https://github.com/dcnieho/Titta/commit/9ed29e48efe9949141a1b77224341355d0421207. Note also that you should remove the call to `vp.start()` on the next line, i have a different solution. Of course use the very latest version of Titta.

Can this issue hereby be closed?

paul-gallenkemper, 2024-12-02:
Yes! This works for our setup and can be resolved :)

closed 2024-12-02 by paul-gallenkemper