dcnieho/Titta #110: Something went wrong when running demo_experiments to calibrating step

lyxose, 2024-12-19:
I am new to Titta. I am trying to run the demo "readme.m" and "readmeTwoScreens.m" to understand how the Titta works. But both of the scripts raised the same error when it comes to the line of `tobii.calVal{1}         = EThndl.calibrate(wpnt)`, and the error information is shown below:

```
错误使用 Screen
Usage:

[windowPtr,rect]=Screen('OpenOffscreenWindow',windowPtrOrScreenNumber [,color] [,rect] [,pixelSize] [,specialFlags] [,multiSample]);

出错 DrawFormattedText2 (第 663 行)
    [tex.number,tex.rect] = Screen('OpenOffscreenWindow', win, [0 0 0 0], [0 0 RectWidth(drawRect) RectHeight(drawRect)]);

出错 Titta/getTextCache (第 3060 行)
                [~,~,txtbounds,cache] = DrawFormattedText2(text,'win',wpnt,'cacheOnly',true,args{:});

出错 Titta>@(varargin)obj.getTextCache(varargin{:}) (第 2610 行)
            funs    = struct('textCacheGetter',@obj.getTextCache, 'textCacheDrawer', @obj.drawCachedText, 'cacheOffSetter', @obj.positionButtonText, 'colorGetter', @(clr) obj.getColorForWindow(clr,wpnt(end)));

出错 PTBButton/setupButton (第 147 行)
                [setup.tCache{p},tRect(p,:)] = this.funs.textCacheGetter(wpnt,sprintf('<color=%s>%s',clr2hex(setup.textColor{p}),setup.string{p}));

出错 PTBButton (第 57 行)
                this.setupButton(this.wpnt,setup);

出错 Titta/showHeadPositioning (第 2612 行)
            but(2)  = PTBButton(obj.settings.UI.button.setup.toggEyeIm,       qHasEyeIm       , wpnt(end), funs, obj.settings.UI.button.margins);

出错 Titta/calibrate (第 1022 行)
                    [out.attempt{kCal}.setupStatus,qCalReset] = obj.showHeadPositioning(wpnt,out);

出错 readme (第 128 行)
        tobii.calVal{1}         = EThndl.calibrate(wpnt);
```
I am not sure how to fix that.

dcnieho, 2024-12-19:
>             funs    = ***@***.***, 'textCacheDrawer', @obj.drawCachedText, 'cacheOffSetter', @obj.positionButtonText, 'colorGetter', @(clr) obj.getColorForWindow(clr,wpnt(end)));

Thanks for your interest in Titta. The actual error message is not in what
you copied, it would probably be in the terminal above the bits you copied.
Could you send that to me?

lyxose, 2024-12-20:
Thanks for your attention. 
Here is the complete output in my terminal. I checked this information again as well. Could that be a result of SYNCHRONIZATION FAILURE?  But I have already skipped the tests by adding the command `Screen('Preference', 'SkipSyncTests', 1)` at the top of the main script `readme.m`. 
```
>> readme
PTB-INFO: Multi-display setup in explicit multi-display mode detected. Using the following mapping:
PTB-INFO: Screen 0 corresponds to the full Windows desktop area. Useful for stereo presentations in stereomode=4 ...
PTB-INFO: Screen 1 corresponds to the display area of the monitor with the Windows-internal name \\.\DISPLAY1 ...
PTB-INFO: Screen 2 corresponds to the display area of the monitor with the Windows-internal name \\.\DISPLAY2 ...

PTB-INFO: Your version of Matlab 64-Bit is global system DPI aware. On Windows-8 or later, fullscreen onscreen windows will only work 
PTB-INFO: properly timing-wise when displayed on displays with the same pixel density as your systems primary display monitor.
PTB-INFO: For your multi-display setup the stimulus display monitor must have a DPI of (192, 192), matching that of
PTB-INFO: your primary display monitor. Ideally you will only display on the primary display in the first place.
PTB-INFO: Displaying on anything with a different DPI will cause mysterious visual timing problems, sync failures etc.
PTB-INFO: Read 'help RetinaDisplay' for more info on this topic.
PTB-WARNING: Call to PsychAvSetMmMaxThreadCharacteristics() for Vista-MMCSS scheduling failed for threadhandle 00007FFCE0C25900. Setting thread priority to HIGHEST as a work-around...

PTB-WARNING: Couldn't determine end-line of vertical blanking interval for your display! Trouble with beamposition queries?!?
PTB-WARNING: Detected end-line is 1079, which is either lower or more than 1.250000 times higher than vbl startline 2160 --> Out of sane range!
PTB-WARNING: Call to PsychAvSetMmMaxThreadCharacteristics() for Vista-MMCSS scheduling failed for threadhandle 00007FFCE0C25900. Setting thread priority to HIGHEST as a work-around...
PTB-INFO: ==============================================================================================================================
PTB-INFO: WINDOWS DWM DESKTOP COMPOSITOR IS ACTIVE. On this Windows-10 or later system, Psychtoolbox can no longer reliably detect if
PTB-INFO: this will cause trouble for timing and integrity of visual stimuli or not. You might be just fine, or you could be in trouble.
PTB-INFO: Use external measurement equipment and independent procedures to verify reliability of timing if you care about proper timing.
PTB-INFO: ==============================================================================================================================

WARNING: Couldn't compute a reliable estimate of monitor refresh interval! Trouble with VBL syncing?!?

WARNING: Mismatch between measured monitor refresh interval and interval reported by operating system.
This indicates massive problems with VBL sync.

----- ! PTB - ERROR: SYNCHRONIZATION FAILURE ! -----

One or more internal checks (see Warnings above) indicate that synchronization
of Psychtoolbox to the vertical retrace (VBL) is not working on your setup.

This will seriously impair proper stimulus presentation and stimulus presentation timing!
Please read 'help SyncTrouble' for information about how to solve or work-around the problem.
You can force Psychtoolbox to continue, despite the severe problems, by adding the command
Screen('Preference', 'SkipSyncTests', 1); at the top of your script, if you really know what you are doing.

----- ! PTB - WARNING: SYNCHRONIZATION TROUBLE ! -----

One or more internal checks (see Warnings above) indicate that
queries of rasterbeam position are not properly working for your setup.

Psychtoolbox will work around this by using a different timing algorithm, 
but it will cause Screen('Flip') to report less accurate/robust timestamps
for stimulus timing.
Read 'help BeampositionQueries' for more info and troubleshooting tips.

PTB-WARNING: Call to PsychAvSetMmMaxThreadCharacteristics() for Vista-MMCSS scheduling failed for threadhandle 00007FFCE0C25900. Setting thread priority to HIGHEST as a work-around...
4468201291: START SETUP (both eyes)
PTB-DEBUG: DrawText: Failed to load external drawtext plugin [Unknown error].
PTB-INFO: DrawText: Failed to load external drawtext plugin 'libptbdrawtext_ftgl64.dll'. Reverting to legacy GDI text renderer. 'help DrawTextPlugin' for troubleshooting.
Error in function OpenOffscreenWindow: 	Usage error
Invalid rect value provided: Empty rects are not allowed.

WARNING: This session of your experiment was run by you with the setting Screen('Preference', 'SkipSyncTests', 1).
WARNING: This means that some internal self-tests and calibrations were skipped. Your stimulus presentation timing
WARNING: may have been wrong. This is fine for development and debugging of your experiment, but for running the real
WARNING: study, please make sure to set Screen('Preference', 'SkipSyncTests', 0) for maximum accuracy and reliability.
错误使用 Screen
Usage:

[windowPtr,rect]=Screen('OpenOffscreenWindow',windowPtrOrScreenNumber [,color] [,rect] [,pixelSize] [,specialFlags] [,multiSample]);

出错 DrawFormattedText2 (第 663 行)
    [tex.number,tex.rect] = Screen('OpenOffscreenWindow', win, [0 0 0 0], [0 0 RectWidth(drawRect) RectHeight(drawRect)]);

出错 Titta/getTextCache (第 3060 行)
[... cut ...]
                [setup.tCache{p},tRect(p,:)] = this.funs.textCacheGetter(wpnt,sprintf('<color=%s>%s',clr2hex(setup.textColor{p}),setup.string{p}));

出错 PTBButton (第 57 行)
                this.setupButton(this.wpnt,setup);

出错 Titta/showHeadPositioning (第 2612 行)
            but(2)  = PTBButton(obj.settings.UI.button.setup.toggEyeIm,       qHasEyeIm       , wpnt(end), funs, obj.settings.UI.button.margins);

出错 Titta/calibrate (第 1022 行)
                    [out.attempt{kCal}.setupStatus,qCalReset] = obj.showHeadPositioning(wpnt,out);

出错 readme (第 128 行)
        tobii.calVal{1}         = EThndl.calibrate(wpnt);
```

iandol, 2024-12-20:
For some reason the rect is not passed, can you put a dbstop at line 663 to see what values are passed to `Screen('OpenOffscreenWindow'...`

```
Error in function OpenOffscreenWindow: 	Usage error
Invalid rect value provided: Empty rects are not allowed.
```

There was also a drawtext error, you should follow the `help DrawTextPlugin` troubleshooting.

```
DrawText: Failed to load external drawtext plugin 'libptbdrawtext_ftgl64.dll'. Reverting to legacy GDI text renderer. 'help DrawTextPlugin' for troubleshooting.
```

dcnieho, 2024-12-20:
Thanks Ian! Yeah, the problem is that drawtext plugin is not working,
please fix that. My code should fall back to another branch in this case,
but apparently it didn't detect thatthe high quality text renderer doesn't
work. Anyway, making that work will fix this (and give you a proper-looking
interface)

lyxose, 2024-12-20:
That explains a lot. I will try to fix the problem with the drawtext plugin. 
But I have already installed the `gstreamer-1.22.5` with the `3.0.19.0` version of PTB, in the platform of WIN11, MATLAB R2022b. It seems that I ran into the problem mentioned in issue [#583](https://github.com/Psychtoolbox-3/Psychtoolbox-3/issues/583). I am still looking for a solution. Thank you very much for your help.

lyxose, 2024-12-29:
The problem was solved by updating psychtoolbox to version `3.0.19.16` with `gstreamer-1.22.5`. Thanks for all the help.

closed 2024-12-29 by lyxose

dcnieho, 2025-01-08:
I am glad that you were able to fix the problem, and thank you for posting the solution!

dcnieho, 2025-03-10:
Hi Martina,

This is a very different problem. But no, it is no problem to add this line for the functioning of Titta.

All the best,
Dee