dcnieho/Titta #111: How to get the real time head position?

lyxose, 2025-01-06:
In the head positioning procedure, Titta draws a line with a changing number at the head of the screen, noting the current distance between the head and the screen. But I found that the position of the head is obtained by private functions `showHeadPositioning` and `setupHead`. I am just wondering if there is another more direct and simple way to get the head position whenever I want.

dcnieho, 2025-01-07:
@lyxose Thanks for your interest! The head position info  (or to be exact, eyeball position) is included in a gaze sample. So you can get it directly in the same way you get gaze (see demos)

lyxose, 2025-01-08:
Thank you so much! That is exactly the way!

closed 2025-01-08 by lyxose